Mellanox/mstflint #4: Unsupported device error

Electronickss, 2017-08-30:
Card: connectX-2 EN

I know i'm using a device that is past what MFT and I guess mst supports but what version of this software do I need to use that does. I need to turn off pxe booting on this card

adham-mellanox, 2017-10-29:
Can you please supply the command line you ran and the output you got ?

ajs124, 2017-11-24:
Same problem here. Command: `mstconfig -e -d 06:00.0 q`. Output: `-E- Unsupported device`.

I mean, mstconfig does say it only supports "ConnectX3, ConnectX3-Pro (FW 2.31.5000 and above)", but what are we supposed to use with ConnectX2? Is there an old version available or something?

adham-mellanox, 2017-11-26:
ConnectX-2 is not supported under mstconfig.
Need to locate older version under openfabrics last one that supports ConnectX2 is 3.8.0: https://www.openfabrics.org/downloads/mstflint/

closed 2017-11-26 by adham-mellanox

ajs124, 2017-11-26:
Cool, that's all I wanted to know. Thanks a lot for your support!

Dante4, 2021-10-03:
Fun fact, you can control ConnectX-2 just fine under windows 10, but not in Linux-based distributive

nospam2000, 2025-04-01:
I just uploaded the sourcecode for mstflint [here](https://github.com/nospam2000/mstflint380) and fixed some compile errors so it now compiles with Debian bookworm and trixie.