Mellanox/mstflint #5: -Werror=implicit-fallthrough gcc 7.1.1 (Arch Linux)

haata, 2017-09-06:
## Error

```
[admin@MasMac mstflint]$ make -j
make  all-recursive
make[1]: Entering directory '/home/admin/git/mstflint'
Making all in common
make[2]: Entering directory '/home/admin/git/mstflint/common'
sed -e 's,[@]MFT_VERSION_STR[@],mstflint 4.7.0,g' -e 's,[@]TOOLS_BUILD_TIME[@],Sep 06 2017\, 15:43:52,g' -e 's,[@]TOOLS_GIT_SHA[@],N/A,g' < ./tools_version.py.in > tools_version.py
sed -e 's,[@]MST_LIB_DIR[@],/usr/local/lib,g' -e 's,[@]MST_BIN_DIR[@],/usr/local/bin,g' -e 's,[@]MFTCONF_PREFIX[@],/usr/local,g' < ./python_wrapper.sh > python_wrapper
chmod +x python_wrapper
make[2]: Leaving directory '/home/admin/git/mstflint/common'
Making all in mft_utils
make[2]: Entering directory '/home/admin/git/mstflint/mft_utils'
gcc -DHAVE_CONFIG_H -I. -I..  -I. -I../common -fPIC  -MD -pipe -Wall -W -Werror -fPIC -g -O2 -Werror -DMST_UL -MT mft_sig_handler.o -MD -MP -MF .deps/mft_sig_handler.Tpo -c -o mft_sig_handler.o mft_sig_handler.c
g++ -DHAVE_CONFIG_H -I. -I..  -I. -I../common -fPIC   -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT errmsg.o -MD -MP -MF .deps/errmsg.Tpo -c -o errmsg.o errmsg.cpp
gcc -DHAVE_CONFIG_H -I. -I..  -I. -I../common -fPIC  -MD -pipe -Wall -W -Werror -fPIC -g -O2 -Werror -DMST_UL -MT calc_hw_crc.o -MD -MP -MF .deps/calc_hw_crc.Tpo -c -o calc_hw_crc.o calc_hw_crc.c
g++ -DHAVE_CONFIG_H -I. -I..  -I. -I../common -fPIC   -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT mlarge_buffer.o -MD -MP -MF .deps/mlarge_buffer.Tpo -c -o mlarge_buffer.o mlarge_buffer.cpp
mv -f .deps/calc_hw_crc.Tpo .deps/calc_hw_crc.Po
mv -f .deps/mft_sig_handler.Tpo .deps/mft_sig_handler.Po
mv -f .deps/mlarge_buffer.Tpo .deps/mlarge_buffer.Po
mv -f .deps/errmsg.Tpo .deps/errmsg.Po
rm -f libmftutils.a
ar cru libmftutils.a mft_sig_handler.o errmsg.o calc_hw_crc.o mlarge_buffer.o 
ar: `u' modifier ignored since `D' is the default (see `U')
ranlib libmftutils.a
make[2]: Leaving directory '/home/admin/git/mstflint/mft_utils'
Making all in ext_libs
make[2]: Entering directory '/home/admin/git/mstflint/ext_libs'
Making all in muparser
make[3]: Entering directory '/home/admin/git/mstflint/ext_libs/muparser'
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParserBase.o -MD -MP -MF .deps/muParserBase.Tpo -c -o muParserBase.o muParserBase.cpp
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParserBytecode.o -MD -MP -MF .deps/muParserBytecode.Tpo -c -o muParserBytecode.o muParserBytecode.cpp
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParserCallback.o -MD -MP -MF .deps/muParserCallback.Tpo -c -o muParserCallback.o muParserCallback.cpp
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParser.o -MD -MP -MF .deps/muParser.Tpo -c -o muParser.o muParser.cpp
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParserError.o -MD -MP -MF .deps/muParserError.Tpo -c -o muParserError.o muParserError.cpp
g++ -DHAVE_CONFIG_H -I. -I../..    -W -Werror -g -MP -MD -fPIC -Wno-deprecated-declarations -g -O2 -DHAVE_TERMIOS_H -Werror -DMST_UL -MT muParserTokenReader.o -MD -MP -MF .deps/muParserTokenReader.Tpo -c -o muParserTokenReader.o muParserTokenReader.cpp
mv -f .deps/muParserCallback.Tpo .deps/muParserCallback.Po
muParserBase.cpp: In member function ‘void mu::ParserBase::CheckOprt(const string_type&, const mu::ParserCallback&, const string_type&) const’:
muParserBase.cpp:379:33: error: this statement may fall through [-Werror=implicit-fallthrough=]
       case cmOPRT_POSTFIX: Error(ecINVALID_POSTFIX_IDENT, -1, a_sName);
                            ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
muParserBase.cpp:380:7: note: here
       case cmOPRT_INFIX:   Error(ecINVALID_INFIX_IDENT, -1, a_sName);
       ^~~~
muParserBase.cpp:380:33: error: this statement may fall through [-Werror=implicit-fallthrough=]
       case cmOPRT_INFIX:   Error(ecINVALID_INFIX_IDENT, -1, a_sName);
                            ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
muParserBase.cpp:381:7: note: here
       default:             Error(ecINVALID_NAME, -1, a_sName);
       ^~~~~~~
muParserBase.cpp: In member function ‘void mu::ParserBase::CreateRPN() const’:
muParserBase.cpp:1230:17: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 ++stArgCount.top();
                 ^~~~~~~~~~~~~~~~~~
muParserBase.cpp:1233:9: note: here
         case cmEND:
         ^~~~
muParserBase.cpp:1288:33: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 m_nIfElseCounter++;
                 ~~~~~~~~~~~~~~~~^~
muParserBase.cpp:1291:9: note: here
         case cmLAND:
         ^~~~
mv -f .deps/muParser.Tpo .deps/muParser.Po
mv -f .deps/muParserError.Tpo .deps/muParserError.Po
mv -f .deps/muParserBytecode.Tpo .deps/muParserBytecode.Po
mv -f .deps/muParserTokenReader.Tpo .deps/muParserTokenReader.Po
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:427: muParserBase.o] Error 1
[... 87 lines cut ...]
mtcr_ul_com.c:1351:46: error: ‘%s’ directive writing up to 255 bytes into a region of size 43 [-Werror=format-overflow=]
         sprintf(fname, "/sys/bus/pci/devices/%s/vendor", dir->d_name);
                                              ^~
mtcr_ul_com.c:1351:9: note: ‘sprintf’ output between 29 and 284 bytes into a destination of size 64
         sprintf(fname, "/sys/bus/pci/devices/%s/vendor", dir->d_name);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
mtcr_ul_com.c: In function ‘reopen_pci_as_inband’:
mtcr_ul_com.c:2125:33: error: ‘%s’ directive writing up to 255 bytes into a region of size 127 [-Werror=format-overflow=]
         sprintf(subdirname, "%s/%s/device", dirname, dir->d_name);
                                 ^~
mtcr_ul_com.c:2125:9: note: ‘sprintf’ output 9 or more bytes (assuming 264) into a destination of size 128
         sprintf(subdirname, "%s/%s/device", dirname, dir->d_name);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
mtcr_ul_com.c:2138:41: error: ‘%s’ directive writing up to 255 bytes into a region of size 121 [-Werror=format-overflow=]
             sprintf(inband_dev, "ibdr-0,%s,1", dir->d_name);
                                         ^~
mtcr_ul_com.c:2138:13: note: ‘sprintf’ output between 10 and 265 bytes into a destination of size 128
             sprintf(inband_dev, "ibdr-0,%s,1", dir->d_name);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

## Notes

There are some more problems with gcc 7.1.1, but these are the first bunch.

adrianchiris, 2017-10-29:
Issue was fixed with using -Wno-implicit-fallthrough

adham-mellanox, 2017-10-29:
Issue was fixed with using -Wno-implicit-fallthrough

closed 2017-10-29 by adham-mellanox