Mellanox/mstflint #6: Can't compile from tarball because gitversion.h is missing

ajs124, 2017-11-22:
If you download the tarball from the release website and try `./autogen.sh && ./configure && make` the build fails because it is not running in a git repo and cannot generate gitversion.h

adham-mellanox, 2017-11-26:
Which tarball are you downloading ?
You should download this one: mstflint-4.8.0.tar.gz (https://github.com/Mellanox/mstflint/files/1457191/mstflint-4.8.0.tar.gz), and rebuild it with rpm --rebuild, or extract and run configure directly.
If you are downloading the source code, and running autogen.sh, it requires that git is installed in your machine to generate the missing file.

ajs124, 2017-11-26:
Seems like I just chose the wrong tarball. After running autogen.sh, which complains like this, even with the right tar:
```
fatal: Not a git repository (or any parent up to mount point /files)                                                                                                                                                                                                              
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).                                                                                                                                                                                                        
```
And running `./configure`, `make` fails like:
```
g++ -DHAVE_CONFIG_H -I. -I..  -I. -I.. -I../include/mtcr_ul -I../mtcr_ul -I../common  -I../tools_layouts -I../mft_utils -I../dev_mgt -I../cmdif -fPIC   -Wall -W -g -MP -MD -pipe -Wno-deprecated-declarations -fPIC -DDATA_PATH=\"/usr/local/share/mstflint\" -DDISABLE_XML2 -DNO_OPEN_SSL -DMLXCFG_NAME=\"mstconfig\" -DMST_DEV_EXAMPLE=\"04:00.0\" -DMST_DEV_EXAMPLE2=\"05:00.0\" -fPIC -fstack-protector -pipe -DHAVE_TERMIOS_H -Werror -DMST_UL -Wno-implicit-fallthrough -Wno-format-overflow -MT mlxcfg_parser.o -MD -MP -MF .deps/mlxcfg_parser.Tpo -c -o mlxcfg_parser.o mlxcfg_parser.cpp
In file included from mlxcfg_parser.cpp:41:0:
../common/tools_version.h:45:10: fatal error: gitversion.h: No such file or directory
 #include "gitversion.h"
          ^~~~~~~~~~~~~~
compilation terminated.
```

That is, with [this](https://github.com/Mellanox/mstflint/archive/v4.8.0-2.tar.gz) tarball, not [that](https://github.com/Mellanox/mstflint/files/1457191/mstflint-4.8.0.tar.gz).

I thought that the proper tarball was just binaries, so I chose the "normal" github packaged tarball.

closed 2017-11-26 by ajs124

jamespharvey20, 2018-10-16:
> If you are downloading the source code, and running autogen.sh, it requires that git is installed in your machine to generate the missing file.

I could be wrong, but I don't see how the "Source code (tar.gz)" releases can ever be used in any circumstance.

I think not only git is required to be installed, but also that `autogen.sh` be ran within the mstflint git tree.

If it's ran within an extracted source tarball (i.e. https://github.com/Mellanox/mstflint/archive/v4.10.0-3.tar.gz), `gitversion.sh:3` runs `if git rev-parse --short HEAD > gitversion.tmp` which: prints `fatal: not a git repository (or any parent up to mount point /)`; and more importantly doesn't create a `common/gitversion.h` which causes a compilation failure in `mlxcfg_parser.cpp`.

Trying to use https://github.com/Mellanox/mstflint/releases/download/v4.10.0-3/mstflint-4.10.0-3.tar.gz fails for me, because aclocal is triggered, and it's expecting to run `aclocal-1.13`.  I'm on Arch, which instead has `aclocal-1.16`.  (My knowledge of autotools is barely enough to get by.  Maybe there's a way around this, but I'm assuming I need to be running `autogen.sh` because of this.)

On a distribution without aclocal-1.13, you either have to manually create `common/gitversion.h` after `autogen.sh` fails, or not use the "Source code (tar.gz)" but instead use a git clone.