Mellanox/mstflint #8: [regression issue] mstvpd does not work over mlx5 device

HonggangLI, 2017-12-04:
Hi,

I built and installed latest mstflint. It seems there is a regression issue for mstvpd. I ran
'git bisect' to narrow down the bad commit.

mstflint]# lspci | grep -i mellanox
04:00.0 Infiniband controller: Mellanox Technologies MT27600 [Connect-IB]

mstflint]# /opt/mstflint-v4.7.0-1/bin/mstvpd 04:00.0
ID: CB194A - Connect-IB QSFP
PN: MCB194A-FCAT         
EC: A9
SN: MT1340X00409            
V0: PCIe Gen3 x16 

mstflint]# /opt/mstflint-v4.8.0-2/bin/mstvpd 04:00.0
-E- Failed to read VPD from 04:00.0!

mstflint]# git bisect good
4201e56ec8e8f8b0767208ba2e1009e76b54bceb is the first bad commit
commit 4201e56ec8e8f8b0767208ba2e1009e76b54bceb
Author: Idan Mehalel <idanme@mellanox.com>
Date:   Wed Oct 18 16:51:34 2017 +0300

    Title: use mvpd in mstvpd
    
    Description:
    Issue: 1081769

HonggangLI, 2017-12-07:
mstvpd/main -> mvpd_get_raw_vpd -> my_vpd_read -> mvpd_read4 -> mvpd_read4_ul -> mvpd_read4_ul_int -> fread(/sys/bus/pci/devices/%04x:%02x:%02x.%d/vpd)

This function call chain try to read VPD_MAX_SIZE (4096) bytes from the vpd file.  fread() does not distinguish between end-of-file and error. So, when the vpd file can't provide 4096 bytes and fread() reach the end of vpd file, fread() return a non-zero code.

We should check with feof() when fread() return an error. This patch fixes this issue.

diff --git a/mtcr_ul/mtcr_ul_com.c b/mtcr_ul/mtcr_ul_com.c
index ddafd0a..9f328b5 100644
--- a/mtcr_ul/mtcr_ul_com.c
+++ b/mtcr_ul/mtcr_ul_com.c
@@ -2575,9 +2575,11 @@ int mvpd_read4_ul_int(mfile *mf, unsigned int offset, u_int8_t value[4]){
         return errno;
     }
     if (fread(value, 1, 4, f) < 1) {
-        //fprintf(stderr, "Failed to read from (%s): %s\n", proc_dev, strerror(errno));
-        fclose(f);
-        return errno;
+        if (!feof(f)) {
+            //fprintf(stderr, "Failed to read from (%s): %s\n", proc_dev, strerror(errno));
+            fclose(f);
+            return errno;
+        }
     }
     fclose(f);
     return 0;

adham-mellanox, 2018-02-26:
@HonggangLI Do you still have the issue ?

HonggangLI, 2018-03-01:
Confirmed that issue had been fixed. Thanks

closed 2018-03-01 by HonggangLI