Mellanox/mstflint #12: configure: EL6 compiler warnings

aron-silverton, 2018-01-03:
On a clean install of EL6:

./configure: line 17373: syntax error near unexpected token `-Wno-implicit-fallthrough,'
./configure: line 17373: `AX_CHECK_COMPILE_FLAG(-Wno-implicit-fallthrough, CXXFLAGS="$CXXFLAGS -Wno-implicit-fallthrough", , )'

$ gcc --version
gcc (GCC) 4.4.7 20120313 (Red Hat 4.4.7-18)

mehdid, 2018-01-05:
FWIW, I've resolved this issue in Debian using the following patch:

https://sources.debian.org/patches/mstflint/4.8.0-1/0011-Declare-m4-as-an-AC_CONFIG_MACRO_DIR-in-configure.ac.patch/

adham-mellanox, 2018-02-22:
@mehdid the link you provided is broken, can you please share it again ?

mehdid, 2018-02-22:
Here it is: https://sources.debian.org/patches/mstflint/4.8.0-2/0011-Declare-m4-as-an-AC_CONFIG_MACRO_DIR-in-configure.ac.patch/

From https://sources.debian.org/patches/mstflint/, you have access to all versions of mstflint available in all Debian releases. Picking the newest one will show you all applied patches.

adham-mellanox, 2018-02-26:
@aron-silverton what is the autotools version do you have ?

aron-silverton, 2018-03-13:
Name    : autoconf
Arch    : noarch
Version : 2.69
Release : 11.el7

Name    : automake
Arch    : noarch
Version : 1.13.4
Release : 3.el7

Name    : libtool
Arch    : x86_64
Version : 2.4.2
Release : 22.el7_3

Let me know if you need more info.

aron-silverton, 2018-03-13:
@adham-mellanox, I'm trying to recreate - I don't currently have access to an EL6 machine.

aron-silverton, 2018-05-11:
@adham-mellanox, any thoughts? I also have to run autoreconf --install to get configure to run. Running autogen.sh is not sufficient.

aron-silverton, 2018-05-11:
@mehdid, I believe that "autoreconf --install" solved the undefined macro issue for me. It does not fix the configure errors:

./configure: line 17536: syntax error near unexpected token `-Wno-implicit-fallthrough,'
./configure: line 17536: `AX_CHECK_COMPILE_FLAG(-Wno-implicit-fallthrough, CXXFLAGS="$CXXFLAGS -Wno-implicit-fallthrough", , )'

mehdid, 2018-05-11:
I don't have those errors. Did you try out my fix?

aron-silverton, 2018-05-11:
@mehdid, on EL7, your fix takes care of things. Well, a slight change to your fix. You had:

AC_CONFIG_MACRO_DIRS([m4])

and I used:

AC_CONFIG_MACRO_DIR([m4])

"DIR" instead of "DIRS".

EL7 does not have the configure errors mentioned above. 

EL6 has those errors, and many others. Some of them are probably because @adham-mellanox is expecting Autconf v2.64 or later and we only have v2.6.3 on OL6/EL6.

tzafrir-mellanox, 2020-02-24:
Is there still need for this fix? FWIW, we build packages on RHEL6 internally, but run the configure elsewhere.

aron-silverton, 2020-02-26:
There may be a workaround using the "autoconf-archive" package in the build environment but it needs to be tested.

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540