Mellanox/mstflint #18: Can't compile mstflint-4.9 with gcc-8

HonggangLI, 2018-03-01:
Failed to compile mstflint-4.9 with Fedora-28 or Fedora-Rawhide (F29). gcc-8 terminated because "-W-error=cast-function-type" and "-W-error=class-memaccess".

export CFLAGS="$RPM_OPT_FLAGS -Wno-error=cast-function-type"
export CXXFLAGS="$RPM_OPT_FLAGS -std=gnu++98 -Wno-c++11-compat -Wno-error=cast-function-type -Wno-error=class-memaccess"

With those two env parameters, compiling works again. But it is just a workaround, please consider to fix those gcc/g++ syntax error.

thanks

wuxb45, 2018-06-21:
+1

This is what I added to make it build:
```
  CXXFLAGS='-O3 -Wno-cast-function-type -Wno-maybe-uninitialized -Wno-format-truncation -Wno-class-memaccess'\
  CFLAGS='-O3 -Wno-cast-function-type -Wno-maybe-uninitialized -Wno-format-truncation'\
  LDFLAGS='-O3 -flto' ./configure --prefix=/usr \
              --enable-fw-mgr
```

HonggangLI, 2018-07-26:
issue had been fixed. thanks

closed 2018-07-26 by HonggangLI