Mellanox/mstflint #22: flint truncating long PCI addresses, device file cannot be opened

ndusek, 2018-06-28:
Hello,

I am trying to upgrade the firmware on some Mellanox ConnectX-5 cards and am running into some issues that I think are related to the flint tool. Here is the system information:

Motherboard: Intel S2600BP
OS: RHEL 7.4
Card: ConnectX-5 VPI (MCX556A-ECAT)
Card FW: 16.20.1010
MFT version: 4.9.0-38
OFED version: 4.3-1.0.1.0-rhel7.4

I can see the cards in the `lspci` output (see below). Note the longer-than-usual PCI addresses (10002:01:00.0/1). This is the first time I have encountered a PCI address like this, but supposedly, on systems with multiple PCI bridges (which this one has) this can happen.

```
10002:01:00.0 Infiniband controller: Mellanox Technologies MT27800 Family [ConnectX-5]
	Subsystem: Mellanox Technologies Device 0001
10002:01:00.1 Infiniband controller: Mellanox Technologies MT27800 Family [ConnectX-5]
	Subsystem: Mellanox Technologies Device 0001
```

I am trying to query the device with flint using the following command:

```
flint -d 10002:01:00.0 q
```
And the following error is being thrown:

```
Failed to open (/sys/bus/pci/devices/0002:01:00.0/config) for reading: No such file or directory
```
Notice that the file it is looking for has had the leading 1 removed, so that instead of 10002:01:00.0 it is looking for 0002:01:00.0 which is not a real device.

I cloned the git repo and tried to find where in the code this might be happening, but I do not have enough knowledge of C++ to make a worthwhile effort.

Thank you

ndusek, 2018-06-28:
Actually, I think I found where this is happening. In **mtcr_ul/mtcr_ul_com.c** PCI domains are limited to a field width of 4. Here is one such instance from lines 423-424:

```
    char mbuf[] = "/sys/bus/pci/devices/XXXX:XX:XX.X/resource";
    sprintf(mbuf, "/sys/bus/pci/devices/%4.4x:%2.2x:%2.2x.%1.1x/resource", domain, bus, dev, func);
```

This is consistent with the define statement at the top of the file:

```
#define DBDF             "%4.4x:%2.2x:%2.2x.%1.1x"
```

I suppose you cannot just change the precision to `%5.5x` because then it would fail for device IDs that have a domain width of 4. So there has to be some sort of input validation at the beginning that checks how long the PCI domain is and then adjusts the device string accordingly.

adham-mellanox, 2018-07-02:
The tool does not support such PCI address format, can you please elaborate more about the setup you have: 
1. OS
2. CPU Arch
If the package should support the OS and the CPU Arch, so a feature request should be opened for next release.

ndusek, 2018-07-02:
Actually, over the weekend, I learned that the 5-digit PCI domains that our machines are reporting do not conform to the PCI standard. So I think it's a problem that we have to work out at the OS level. Therefore, I will go ahead and close this issue, but thank you for responding.

closed 2018-07-02 by ndusek

ahlabenadam, 2019-09-05:
based on man page of lspci, domains number are from 0 to ffff. Seems that having 5 digit in domain is not a standart

alspoov, 2021-10-01:
Check the BIOS configuration of the server this is on.  I had a similar issue where I was using VROC with Intel VMD enabled on too many slots, pushing the cards over to the "dark side" of OS's reach.  If you use Supermicro, you may want to review page 164 of this manual: https://www.supermicro.com/manuals/superserver/1U/MNL-1978.pdf.   

Alex.