Mellanox/mstflint #23: v4.9 added libibverbs dependency

aron-silverton, 2018-06-29:
Hi @adham-mellanox,

It looks like [Porting MFT-4.9.0 changes](https://github.com/Mellanox/mstflint/commit/50fcec407c5f83e4bd5549805ce90c121760497c) added a dependency on libibverbs:

`#if !defined(__WIN__) && !defined(DISABLE_OFED)`
`#include <dlfcn.h>`
`#include <infiniband/verbs.h>`
`#endif`

but there is no BuildRequires in the spec file.

Perhaps:

`BuildRequires: libibverbs-devel` 

which should pull in rdma-core-devel when appropriate since it provides libibverbs-devel in newer distros.

DanGoldbergMellanox, 2018-07-08:
Hi @aron-silverton,
The dependency was removed in the following commit: 40d9d3a (master_devel) and will be merged into master this week.

closed 2018-07-08 by DanGoldbergMellanox

aron-silverton, 2018-07-09:
Thanks, @DanGoldbergMellanox!