Mellanox/mstflint #25: Please link shared libraries with '-Wl,-z,now'.

HonggangLI, 2018-09-27:
annocheck error message:
Hardened: ./usr/lib64/mstflint/python_tools/c_dev_mgt.so: FAIL: Not linked with -Wl,-z,now.
Hardened: ./usr/lib64/mstflint/python_tools/ccmdif.so: FAIL: Not linked with -Wl,-z,now.
Hardened: ./usr/lib64/mstflint/python_tools/cmtcr.so: FAIL: Not linked with -Wl,-z,now.
Hardened: ./usr/lib64/mstflint/python_tools/rreg_access.so: FAIL: Not linked with -Wl,-z,now.

```
diff --git a/cmdif/Makefile.am b/cmdif/Makefile.am
index d4f89a2..46604f5 100644
--- a/cmdif/Makefile.am
+++ b/cmdif/Makefile.am
@@ -46,7 +46,7 @@ libcmdif_a_SOURCES = tools_cif.c tools_cif.h icmd_cif_common.c icmd_cif_common.h
 cmdif_pylibdir = $(libdir)/mstflint/python_tools/
 cmdif_pylib_DATA = ${CCMDIF_SO} cmdif.py
 ${CCMDIF_SO}: libcmdif.a
-	$(CC) -g -Wall -pthread -shared ${CFLAGS} *.o -o ${CCMDIF_SO} \
+	$(CC) -g -Wall -pthread -shared ${CFLAGS} ${LDFLAGS} *.o -o ${CCMDIF_SO} \
 	-L$(USER_DIR)/tools_layouts -ltools_layouts -L$(USER_DIR)/${MTCR_CONF_DIR} -lmtcr_ul
 
 CLEANFILES = ${CCMDIF_SO}
diff --git a/dev_mgt/Makefile.am b/dev_mgt/Makefile.am
index 907e5ce..a7e2249 100644
--- a/dev_mgt/Makefile.am
+++ b/dev_mgt/Makefile.am
@@ -50,7 +50,7 @@ dev_mgt_pylib_DATA = c_dev_mgt.so dev_mgt.py
 dist_dev_mgt_pylib_DATA = dev_mgt.py
 
 c_dev_mgt.so: libdev_mgt.a
-	$(CC) -g -Wall -pthread -shared ${CFLAGS} tools_dev_types.o -o c_dev_mgt.so \
+	$(CC) -g -Wall -pthread -shared ${CFLAGS} ${LDFLAGS} tools_dev_types.o -o c_dev_mgt.so \
 	-L$(USER_DIR)/reg_access -lreg_access \
 	-L$(USER_DIR)/tools_layouts -ltools_layouts \
 	-L$(USER_DIR)/${MTCR_CONF_DIR} -lmtcr_ul
diff --git a/mtcr_py/Makefile.am b/mtcr_py/Makefile.am
index 5f5677d..566a870 100644
--- a/mtcr_py/Makefile.am
+++ b/mtcr_py/Makefile.am
@@ -39,7 +39,7 @@ MTCR_DIR = $(USER_DIR)/${MTCR_CONF_DIR}
 mtcr_pylib_DATA = cmtcr.so mtcr.py
 dist_mtcr_pylib_DATA = mtcr.py
 cmtcr.so:
-	$(CC) -g -Wall -pthread -shared ${CFLAGS} $(MTCR_DIR)/*.o -o cmtcr.so
+	$(CC) -g -Wall -pthread -shared ${CFLAGS} ${LDFLAGS} $(MTCR_DIR)/*.o -o cmtcr.so
 
 CLEANFILES = cmtcr.so
 
diff --git a/reg_access/Makefile.am b/reg_access/Makefile.am
index 353fe02..7d70fd8 100644
--- a/reg_access/Makefile.am
+++ b/reg_access/Makefile.am
@@ -53,7 +53,7 @@ reg_access_pylib_DATA = ${RREG_ACCESS_SO} regaccess.py
 dist_reg_access_pylib_DATA = regaccess.py
 
 ${RREG_ACCESS_SO}: libreg_access.a
-	$(CC) -g -Wall -pthread -shared ${CFLAGS} *.o -o ${RREG_ACCESS_SO} \
+	$(CC) -g -Wall -pthread -shared ${CFLAGS} ${LDFLAGS} *.o -o ${RREG_ACCESS_SO} \
 	-L$(USER_DIR)/tools_layouts -ltools_layouts -L$(MTCR_DIR) -lmtcr_ul
 
 CLEANFILES = ${RREG_ACCESS_SO}
```

HonggangLI, 2019-02-01:
Issue persists for rpm build from mstflint-4.11.0-2 .

annocheck mstflint-4.11.0-2.x86_64.rpm  | grep now
Hardened: c_dev_mgt.so: FAIL: Not linked with -Wl,-z,now.
Hardened: ccmdif.so: FAIL: Not linked with -Wl,-z,now.
Hardened: cmtcr.so: FAIL: Not linked with -Wl,-z,now.
Hardened: rreg_access.so: FAIL: Not linked with -Wl,-z,now.

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540