Mellanox/mstflint #33: mstconfig Fails to query device configuration on Connectx-3 EN cards

sotiris-bos, 2019-02-05:
I have two Connectx-3 EN single SFP+ port cards that I am trying to change the config. FW 2.42.5000 / FlexBoot 3.4.752

I have used multiple versions of the OFED driver as well as the EN driver with multiple versions of mstflint.

I have tried Centos 7.6, Fedora 29 and Ubuntu 16.04.

I have reflashed the cards multiple times as well as reset the config. I even flashed them from livefish mdoe.

Nothing seems to work, I can't query the config of the cards with either mstflint or MFT.

davidebeatrici, 2021-07-02:
I'm encountering the issue as well, on Debian Bullseye.

davidebeatrici, 2021-07-05:
I think it's worth mentioning that `ethtool` is unable to retrieve info about the transceiver, after updating the firmware to 2.42.5000.

```
$ sudo ethtool -m enp4s0
Cannot get module EEPROM information: Input/output error
```

```
mlx4_core 0000:04:00.0: MLX4_CMD_MAD_IFC Get Module info attr(ff60) port(1) i2c_addr(50) offset(0) size(2): Response Mad Status(31c) - cable is not connected
```

```
$ sudo ethtool -i enp4s0
driver: mlx4_en
version: 4.0-0
firmware-version: 2.42.5000
expansion-rom-version: 
bus-info: 0000:04:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes
```

`supports-eeprom-access` was set to `yes` with the previous firmware, however many transceiver values were definitely wrong.

rvdm82, 2021-12-24:
I have the same issues with a Connectx-3 EN card - Anyone end up finding a solution ?

davidebeatrici, 2021-12-24:
I now see a different output:

```
$ uname -r
5.14.0-2-amd64
```

```
$ sudo ethtool -m enp4s0
netlink error: Input/output error
```

```
mlx4_core 0000:04:00.0: MLX4_CMD_MAD_IFC Get Module ID attr(ff60) port(1) i2c_addr(50) offset(0) size(1): Response Mad Status(31c) - cable is not connected
```

```
$ sudo ethtool -i enp4s0
driver: mlx4_en
version: 4.0-0
firmware-version: 2.42.5000
expansion-rom-version: 
bus-info: 0000:04:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes
```

```
$ sudo ethtool enp4s0
Settings for enp4s0:
        Supported ports: [ FIBRE ]
        Supported link modes:   1000baseKX/Full
                                10000baseKR/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: No
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseKX/Full
                                10000baseKR/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: 10000Mb/s
        Duplex: Full
        Auto-negotiation: off
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: d
        Wake-on: d
        Current message level: 0x00000014 (20)
                               link ifdown
        Link detected: yes
```

Also, the ~6 seconds delay at boot has not been there anymore for a while now.

closed 2022-01-11 by tomer540