Mellanox/mstflint #35: Please address covscan issues

HonggangLI, 2019-02-06:


HonggangLI, 2019-02-06:
mstflint-4.11.0-2.el8/scan-results-summary.txt

      2	ARRAY_VS_SINGLETON
      1	BAD_COMPARE
      9	BUFFER_SIZE_WARNING
     11	CHECKED_RETURN
     36	CLANG_WARNING
     46	COMPILER_WARNING
      1	CONSTANT_EXPRESSION_RESULT
      2	COPY_PASTE_ERROR
     11	CPPCHECK_WARNING
      5	DC.WEAK_CRYPTO
      8	DEADCODE
     12	FORWARD_NULL
      3	MISSING_BREAK
     15	NEGATIVE_RETURNS
      4	NULL_RETURNS
      5	OVERFLOW_BEFORE_WIDEN
     10	OVERRUN
      8	PASS_BY_VALUE
      4	RETURN_LOCAL
      2	SHELLCHECK_WARNING
      1	SIGN_EXTENSION
      6	SIZEOF_MISMATCH
      3	STRING_OVERFLOW
      1	UNCAUGHT_EXCEPT
      4	UNINIT
     17	UNINIT_CTOR
      1	UNUSED_VALUE

Please download covscan log from this http link:
http://people.redhat.com/honli/mstflint-scan/

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540