Mellanox/mstflint #38: man pages are malformed

bdrung, 2019-02-12:
The man pages in `man/` are malformed, because they use `help2man`. Nearly everything is listed under the DESCRIPTION section. Example:
```
man -l man/mstflint.1
```
I suggest to write the man pages in Markdown and use pandoc to convert them into roff format. Example: ibv_create_counters.3.md in https://github.com/linux-rdma/rdma-core/tree/master/libibverbs/man

tzafrir-mellanox, 2020-02-24:
I can't detect this issue:

for man in ./man/*; do man -l $man >/dev/null; done

(even in the version of master at the time of this commit)

bdrung, 2020-02-25:
Here is what the mstflint.1 from the master branch looks like (skipped a part from the middle):
```
MSTFLINT,(1)                     User Commands                    MSTFLINT,(1)

NAME
       mstflint,  - manual page for mstflint, mstflint 4.13.3, built on Dec 30
       2019, 11:43:54. Git SHA Hash: N/A

DESCRIPTION
       NAME

              MstFlint - Flash Interface

       SYNOPSIS

              MstFlint [OPTIONS] <command> [Parameters]

              [-d|--device <device>] [-i|--image <image>] [--latest_fw] [--ir]
              [-h|--help]  [--hh]  [-y|--yes]  [--no] [--guid <GUID>] [--guids
              <GUIDS...>]  [--mac  <MAC>]  [--macs  <MACs...>]  [--uid  <UID>]
              [--blank_guids]     [--clear_semaphore]    [--qq]    [--low_cpu]
              [--flashed_version]   [--nofs]   [--allow_rom_change]   [--over‐
              ride_cache_replacement]      [--no_flash_verify]      [--use_fw]
              [-s|--silent]  [--vsd  <string>]   [--use_image_ps]   [--use_im‐
              age_guids] [--use_image_rom] [--use_dev_rom] [--ignore_dev_data]
              [--no_fw_ctrl] [--dual_image] [--striped_image] [--banks <bank>]
              [--log     <log_file>]    [--flash_params    <type,    log2size,
              num_of_flashes>]   [-v|--version]   [--private_key   <key_file>]
              [--key_uuid <uuid_file>] [--private_key2 <key_file>] [--hmac_key
              <hmac_key>] [--key_uuid2 <uuid_file>]

       DESCRIPTION

              flint is a FW (firmware) burning  and  flash  memory  operations
              tool for Mellanox Infiniband HCAs,Ethernet NIC cards, and switch
              devices.

       OPTIONS

              MstFlint [OPTIONS] <command> [Parameters]

       -d|--device <device>
              : Device flash is connected to.  Commands affected: all

       -i|--image <image>
              : Binary image file.  Commands affected: burn, verify

       --latest_fw
              : Commands affected: burn

       --ir   : Commands affected: burn

       -h|--help
              : Prints this message and exits

[...]

       image_reactivate|ir
              : Reactivate previous flash image. For FW

              controlled devices only.

       binary_compare|bc
              : Binary compare between device firmware and

              given BIN file. If there is a silent mode, no progress  is  dis‐
              played.

       RETURN VALUES

       0      : Successful completion.

       1      : An error has occurred.

       7      : For burn command - FW already updated - burn

              was aborted.

SEE ALSO
       The full documentation for mstflint, is maintained as a Texinfo manual.
       If the info and mstflint, programs are properly installed at your site,
       the command

              info mstflint,

       should give you access to the complete manual.

mstflint, mstflint 4.13.3, built December020199, 11:43:54. Git SHAMSTFLINT,(1)
```

And here is roughly what I expect:

```
MSTFLINT,(1)                     User Commands                    MSTFLINT,(1)

NAME
       mstflint - Flash Interface

SYNOPSIS
       mstflint [-d|--device <device>] [-i|--image <image>] [--latest_fw] [--ir]
              [-h|--help] [--hh] [-y|--yes] [--no]  [--guid  <GUID>]  [--guids
              <GUIDS...>]  [--mac  <MAC>]  [--macs  <MACs...>]  [--uid  <UID>]
              [--blank_guids]    [--clear_semaphore]    [--qq]     [--low_cpu]
              [--flashed_version]   [--nofs]   [--allow_rom_change]   [--over-
              ride_cache_replacement]      [--no_flash_verify]      [--use_fw]
              [-s|--silent]   [--vsd   <string>]  [--use_image_ps]  [--use_im-
              age_guids] [--use_image_rom] [--use_dev_rom] [--ignore_dev_data]
              [--no_fw_ctrl] [--dual_image] [--striped_image] [--banks <bank>]
              [--log    <log_file>]    [--flash_params    <type,     log2size,
              num_of_flashes>]   [-v|--version]   [--private_key   <key_file>]
              [--key_uuid <uuid_file>] [--private_key2 <key_file>] [--hmac_key
              <hmac_key>] [--key_uuid2 <uuid_file>] <command> [Parameters]

DESCRIPTION

      flint  is  a  FW  (firmware) burning and flash memory operations
      tool for Mellanox Infiniband HCAs,Ethernet NIC cards, and switch
      devices.

OPTIONS
       -d|--device <device>
              Device flash is connected to.  Commands affected: all

       -i|--image <image>
              Binary image file.  Commands affected: burn, verify

       --latest_fw
              : Commands affected: burn

       --ir   : Commands affected: burn

       -h|--help
              Prints this message and exits

       --hh   Prints extended command help

[...]

       --key_uuid2 <uuid_file>
              UUID matching the given private key to be used by the sign  com-
              mand

COMMANDS SUMMARY

       burn|b -ir
              Burn flash. Use "-ir burn" flag to perform

              image reactivation prior burning.

       query|q [full]
              Query misc. flash/firmware characteristics,

              use "full" to get more information.

[...]

       binary_compare|bc
              : Binary compare between device firmware and

              given BIN file. If there is a silent mode, no progress  is  dis-
              played.

RETURN VALUES
       0      Successful completion.

       1      An error has occurred.

       7      For burn command - FW already updated - burn was aborted.

SEE ALSO
       The  full documentation for mstflint is maintained as a Texinfo manual.
       If the info and mstflint programs are properly installed at your  site,
       the command

              info mstflint

       should give you access to the complete manual.

mstflint, mstflint 4.13.3, built December020199, 11:43:54. Git SHAMSTFLINT,(1)
```

See also: https://en.wikipedia.org/wiki/Man_page

tzafrir-mellanox, 2021-02-04:
Update: the Debian package carries a patch that partially fixes this:
https://sources.debian.org/patches/mstflint/4.16.0+1-1/0003-Fix-man-page-errors.patch/

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540

bdrung, 2022-01-12:
A simple `man -l man/mstflint.1` in the master_devel branch shows that the man page is as malformed as reported three years ago. Do you want me to open another issue ticket with the exact same content?

tzafrir-mellanox, 2022-01-16:
Please reopen. Patch seems to apply as-is. Re-submitting bdrung's patch: #509