Mellanox/mstflint #89: Is 4.11.0-4 meant to be tagged pre-release?

jamespharvey20, 2019-04-25:
Very sorry if 4.11.0-4 is meant to be a pre-release.  Not used to seeing that on this project, and without it having a versioning scheme like some projects designating even/odds etc, or a `rc` in the name, wanted to check.  I maintain `mstflint` in the Arch AUR repository, and don't want to release `-4` if it's not meant to be a stable release.  Gut feeling is the wrong box was checked.

jamespharvey20, 2019-06-15:
4.12.0-1 is tagged as a release, but 4.11.0-5 was also tagged as a pre-release.  Are these pre-releases something new, or were they accidentally tagged that way?

tzafrir-mellanox, 2021-02-04:
Is there any need to keep this issue open?

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540