Mellanox/mstflint #181: Spec file has rpmlint issues

aron-silverton, 2019-09-05:
Summarizing email thread...

The RPM produced by the upstream spec file has these rpmlint issues:
```
mstflint.x86_64: W: devel-file-in-non-devel-package /usr/lib64/mstflint/libmtcr_ul.a
mstflint.x86_64: E: script-without-shebang /usr/lib64/mstflint/python_tools/mstfwreset/mstfwreset.py
mstflint.x86_64: W: devel-file-in-non-devel-package /usr/include/mstflint/mtcr.h
mstflint.x86_64: E: script-without-shebang /usr/lib64/mstflint/python_tools/mstprivhost/mstprivhost.py
mstflint.x86_64: W: devel-file-in-non-devel-package /usr/include/mstflint/mtcr_com_defs.h
mstflint.x86_64: E: non-executable-script /usr/lib64/mstflint/python_tools/mstfwtrace/mstfwtrace.py 0644L /usr/bin/env
mstflint.x86_64: W: no-manual-page-for-binary mstfwreset
mstflint.x86_64: W: no-manual-page-for-binary mstcongestion
```
@DanGoldbergMellanox: Regarding devel warnings, we will need to decide if we want to create devel package. It looks as the right thing to do, but it should be checked also with other customers.

@DanGoldbergMellanox: Regarding script without shebang, it’s a common practice in py scripts and it solves py compatibility issues. We don’t access those scripts directly but with a wrapper script. We can have a discussion on preferred way to improve the python packaging process.

@DanGoldbergMellanox: Regarding missing man pages, we should add them.

```
warning: Installed (but unpackaged) file(s) found:
   /usr/bin/mstfwtrace [R] This is the result of the previously mentioned bug (Oracle added the fix for it).
   /usr/include/mstflint/cmdif/icmd_cif_common.h
   /usr/include/mstflint/cmdif/icmd_cif_open.h
   /usr/include/mstflint/common/compatibility.h
   /usr/include/mstflint/mtcr_mf.h
   /usr/include/mstflint/tools_layouts/adb_to_c_utils.h
   /usr/include/mstflint/tools_layouts/icmd_layouts.h
   /usr/lib64/libcmdif.a
   /usr/lib64/libdev_mgt.a
   /usr/lib64/libreg_access.a
   /usr/lib64/libtools_layouts.a
   /usr/lib64/mstflint/python_tools/cmdif.pyc
   /usr/lib64/mstflint/python_tools/cmdif.pyo
   /usr/lib64/mstflint/python_tools/dev_mgt.pyc
   /usr/lib64/mstflint/python_tools/dev_mgt.pyo
   /usr/lib64/mstflint/python_tools/mft_logger.pyc
   /usr/lib64/mstflint/python_tools/mft_logger.pyo
   /usr/lib64/mstflint/python_tools/mlxpci/binary_file.pyc
   /usr/lib64/mstflint/python_tools/mlxpci/binary_file.pyo
   /usr/lib64/mstflint/python_tools/mlxpci/mlxpci_lib.pyc
   /usr/lib64/mstflint/python_tools/mlxpci/mlxpci_lib.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/__init__.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/__init__.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/logger.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/logger.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mcra.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mcra.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlnx_peripheral_components.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlnx_peripheral_components.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_mlnxdriver.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_mlnxdriver.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_status_checker.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_status_checker.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_utils.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/mlxfwreset_utils.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/pci_device.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mlxfwresetlib/pci_device.pyo
   /usr/lib64/mstflint/python_tools/mstfwreset/mstfwreset.pyc
   /usr/lib64/mstflint/python_tools/mstfwreset/mstfwreset.pyo
   /usr/lib64/mstflint/python_tools/mstfwtrace/fw_trace_utilities.pyc
   /usr/lib64/mstflint/python_tools/mstfwtrace/fw_trace_utilities.pyo
   /usr/lib64/mstflint/python_tools/mstfwtrace/mstfwtrace.pyc
   /usr/lib64/mstflint/python_tools/mstfwtrace/mstfwtrace.pyo
   /usr/lib64/mstflint/python_tools/mstfwtrace/secure_fw_trace.pyc
   /usr/lib64/mstflint/python_tools/mstfwtrace/secure_fw_trace.pyo
   /usr/lib64/mstflint/python_tools/mstprivhost/mstprivhost.pyc
   /usr/lib64/mstflint/python_tools/mstprivhost/mstprivhost.pyo
   /usr/lib64/mstflint/python_tools/mtcr.pyc
   /usr/lib64/mstflint/python_tools/mtcr.pyo
   /usr/lib64/mstflint/python_tools/regaccess.pyc
   /usr/lib64/mstflint/python_tools/regaccess.pyo
   /usr/lib64/mstflint/python_tools/tools_version.pyc
   /usr/lib64/mstflint/python_tools/tools_version.pyo
```
Files should either be packaged and included in the RPM or they should not be installed in the RPM BUILDROOT while the RPM is being built. Some of the files listed above probably belong in the missing -devel package if one is ever provided. The binary python files, if built during rpmbuild (i.e., built for the specific platform) should be fine to include in the binary RPM. Any .pyc or .pyo files committed to the git repo for some reason should not be packaged as the files in the repo may not have been built for the correct platform. I think the binary python files should be built as part of rpmbuild or excluded entirely from the RPM.

tzafrir-mellanox, 2020-01-16:
Well, the spec has its issues.

%define optflags -g -O2

aron-silverton, 2020-09-08:
I sent a pull request for the optflags issue back in March: https://github.com/Mellanox/mstflint/pull/313

closed 2022-01-11 by tomer540