Mellanox/mstflint #193: Document FreeBSD device naming

Orum, 2019-09-15:
There's no clear way to actually select a device for usage with these tools on FreeBSD.  It doesn't have 'lspci', it has 'pciconf'--but no permutation that I tried of the card name listed there was usable for the -d parameter.

Here's the example output with an installed card (MNPA19-XTR) on FreeBSD 12.0-RELEASE:
```
# pciconf -l | grep mlx
mlx4_core0@pci0:129:0:0:        class=0x020000 card=0x002115b3 chip=0x675015b3 rev=0xb0 hdr=0x00
```

DanGoldbergMellanox, 2019-09-15:
Hi, please use pci0:129:0:0 as device name for the -d parameter.

Orum, 2019-09-16:
Thanks, that worked.  It'd be nice to see it somewhere in the documentation though, in case someone else has the same problem.

closed 2022-01-11 by tomer540