Mellanox/mstflint #207: default generated configuration script ./configure.sh script fails, missing infiniband/mad.h

thuffmlx, 2019-09-27:
The .configure script generated by ./autogen.sh fails in t he default mode (build with inifiband).

checking infiniband/mad.h usability... no
checking infiniband/mad.h presence... no
checking for infiniband/mad.h... no
configure: error: cannot find infiniband/mad.h . Use --disable-inband to remove this dependaency

This will build successfully when the --disable-inband is used.

But default setup should build without errors.

rob-scheepens, 2020-09-03:
Just hit this again with 4.14.0-3.

thuffmlx, 2020-09-08:
Rob,

You’ll need to set compile flags.

FLAGS="--enable-adb-generic-tools --enable-openssl --disable-inband"

Tom

tomer540, 2022-01-11:
Closed due to time limit, if still require addressing – please open a new ticket.

closed 2022-01-11 by tomer540