Mellanox/mstflint #281: Python scripts with missing or incorrect shebang lines

aron-silverton, 2020-02-07:
With Enterprise Linux 8 distros, Python is either /usr/bin/python2 or /usr/bin/python3 (preferred) and there is no un-versioned /usr/bin/python unless one goes out of their way to make it so.

I find these Python files in mstflint:
```
./cmdif/cmdif.py
./common/mft_logger.py
./common/tools_version.py.in
./dev_mgt/dev_mgt.py
./ext_libs/iniParser/test/twisted-genhuge.py
./mlxconfig/mstprivhost.py
./mtcr_py/mtcr.py
./mtcr_py/test.py
./reg_access/regaccess.py
./small_utils/binary_file.py
./small_utils/mlxfwresetlib/__init__.py
./small_utils/mlxfwresetlib/logger.py
./small_utils/mlxfwresetlib/mcra.py
./small_utils/mlxfwresetlib/mlnx_peripheral_components.py
./small_utils/mlxfwresetlib/mlxfwreset_mlnxdriver.py
./small_utils/mlxfwresetlib/mlxfwreset_status_checker.py
./small_utils/mlxfwresetlib/mlxfwreset_utils.py
./small_utils/mlxfwresetlib/pci_device.py
./small_utils/mlxpci_lib.py
./small_utils/mstfwreset.py
./tracers/fwtrace/fw_trace_utilities.py
./tracers/fwtrace/mstfwtrace.py
./tracers/fwtrace/secure_fw_trace.py
```
I only check files with "py" in their extension and did not look for files that are actually Python but do not include the extension.

Out of those files, I only find one that has a shebang:
```
head -n1 ./tracers/fwtrace/mstfwtrace.py
#!/usr/bin/env python
```

This makes me wonder

1. Are there other Python files that just happen to be missing a .py extension?
2. Should any of the other Python files included a shebang?
3. Are all of the Python files compatible with both python2 and python3 (the ./common/python_wrapper.sh script seems to imply yes...)

If you want to keep a single upstream spec file for both .el7 and .el8 you may want to check "dist" and then modify the Python scripts' shebang lines to be either /usr/bin/python2 or /usr/bin/python3 and conditionally Require and BuildRequires the appropriate Python as needed. If you stick with using GNU autotools, you can probably do this with configure, make, and mstflint.spec.in as well as do the substitution of the shebang lines.

tomer540, 2022-01-13:
1. no, all python scripts follow the convention of .py extention.
2. all python scripts run via wrapper that detect the latest usable version of python to run the script with.
hance no shebang lines are need 

thank you for spotting the error in ./tracers/fwtrace/mstfwtrace.py , we will remove this instance as well

tzafrir-mellanox, 2022-01-16:
Python scripts that should not executed directly from the shell should not be executable. If a script is executed with 'python <path/to/script>, it does not need a shebang line. If those scripts should only be exected with the wrapper, they should not have shebang lines and should not be executable.

They are marked executable:
./resourceparse/Makefile.am:dist_pythonlib_SCRIPTS = mstresourceparse.py

a SCRIPTS type install installs them executable.

closed 2022-05-26 by tomer540