Mellanox/mstflint #310: MST:: : get_space_support_status 438: At least one SPACE is not supported

aron-silverton, 2020-03-17:
It appears that this error message is the result of a test for which address spaces are supported by a device:

```
$ git grep "At least one SPACE is not supported"
kernel/mst_main.c:      mst_err("At least one SPACE is not supported\n");

    if (_set_addr_space(dev, AS_CR_SPACE) || _set_addr_space(dev, AS_ICMD)
            || _set_addr_space(dev, AS_SEMAPHORE)) {
        mst_err("At least one SPACE is not supported\n");
        dev->spaces_support_status = SS_NOT_ALL_SPACES_SUPPORTED;
    } else {
        dev->spaces_support_status = SS_ALL_SPACES_SUPPORTED;
    }
```

As a result of the test with our specific devices, a constant (i.e, SS_NOT_ALL_SPACES_SUPPORTED)is set. But, it does not appear that this constant is ever used elsewhere in the code:

```
$ git grep "SS_NOT_ALL_SPACES_SUPPORTED"
kernel/mst_main.c:  SS_NOT_ALL_SPACES_SUPPORTED = 2
kernel/mst_main.c:      dev->spaces_support_status = SS_NOT_ALL_SPACES_SUPPORTED;
```

If this test is determining something useful about the device then perhaps is should be displayed with a meaningful message with INFO status and not as an error message. If the message is of no use to an operator/admin then it should be surpressed regardless of how it is worded.

I do not think that we've looked at whether the same message is seen with CX-5 operating in Infiniband mode so, at this time, I do not know if this is a CX-3 vs. CX-5 issue and/or an Infiniband vs. RoCE issue. Either way, any messages printed by the driver should be meaningful to the operator.

itayav17, 2020-03-18:
Hi,
This message is intended to report if one of the vendor specific capabilities are not supported for a specific device.
I will change the level to INFO message.

Thanks,
Itay Avraham

aron-silverton, 2020-03-18:
Thanks for the quick response. I still think it would be good if the message said something more meaningful to the end user. Even saying, "device X doesn't support capability Y" would still be confusing to somebody who would then wonder if that is a problem.

itayav17, 2020-03-18:
#312 
You right, I will change it soon.

Thanks,
Itay.

itayav17, 2020-04-21:
https://github.com/Mellanox/mstflint/pull/332
@DanGoldbergMellanox  please review.

closed 2022-01-12 by ahmadawwad