Mellanox/mstflint #324: compilation failed for fedora 33

Honggang-LI, 2020-03-31:
g++ -DHAVE_CONFIG_H -I. -I..  -I.. -I. -I../mtcr_ul -I../mflash -I../common -I../tools_layouts -I../mft_utils -I../include/mtcr_ul  -Wall -W -g -MP -MD -pipe -DEXTERNAL -fPIC   -DFLINT_NAME=\"mstflint\" -DFLINT_DISPLAY_NAME=\"MstFlint\" -DMST_DEV_EXAMPLE1=\"03:00.0\" -DMST_DEV_EXAMPLE2=\"mlx4_0\" -DMST_DEV_EXAMPLE3=\"03:00.0\" -DMST_DEV_EXAMPLE4=\"04:00.0\" -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c -o mstflint-subcommands.o `test -f 'subcommands.cpp' || echo './'`subcommands.cpp
subcommands.cpp: In member function 'bool BurnSubCommand::checkFwVersion(bool, u_int16_t, u_int16_t, u_int16_t)':
subcommands.cpp: In member function 'bool BurnSubCommand::checkFwVersion(bool, u_int16_t, u_int16_t, u_int16_t)':
subcommands.cpp:1852:42: error: format not a string literal and no format arguments [-Werror=format-security]
subcommands.cpp:1852:42: error: format not a string literal and no format arguments [-Werror=format-security]
 1852 |                     false, "N/A").c_str());
 1852 |                     false, "N/A").c_str());
      |                                          ^
      |                                          ^
subcommands.cpp:1862:39: error: format not a string literal and no format arguments [-Werror=format-security]
subcommands.cpp:1862:39: error: format not a string literal and no format arguments [-Werror=format-security]
 1862 |                         "N/A").c_str());
 1862 |                         "N/A").c_str());

Honggang-LI, 2020-03-31:
https://github.com/Mellanox/mstflint/releases/download/v4.14.0-1/mstflint-4.14.0-1.tar.gz

Create a srpm with this tar ball, and run

mock -r fedora-rawhide-x86_64 --rebuild --resultdir=fc33 ./mstflint-4.14.0-1.fc33.src.rpm

It will be terminated with error message in previous comment.

mataneli94, 2020-03-31:
Currently, fedora 33 not supported. Please review the user manual for more info.

Honggang-LI, 2020-03-31:
1004  mock -r fedora-32-x86_64 --rebuild --resultdir=fc32 ./mstflint-4.14.0-1.fc33.src.rpm 
 1005  mock -r fedora-31-x86_64 --rebuild --resultdir=fc31 ./mstflint-4.14.0-1.fc33.src.rpm 

compilation failed for fedora 32 and 31 too.

Honggang-LI, 2020-04-01:
Commit 8e7720685a255 introduced this issue. It can be fixed with this simple patch.

```c
diff --git a/flint/subcommands.cpp b/flint/subcommands.cpp
index 7230f9a59f94..0f6440be7b28 100755
--- a/flint/subcommands.cpp
+++ b/flint/subcommands.cpp
@@ -1847,7 +1847,7 @@ bool BurnSubCommand::checkFwVersion(bool CreateFromImgInfo, u_int16_t fw_ver0, u
     char curr_ver[124], new_ver[124];
     printf("\n");
     printf("    Current FW version on flash:  ");
-    snprintf(curr_ver, 124,
+    snprintf(curr_ver, 124, "%s",
             current.get_fw_version(VERSION_FORMAT(_devInfo.fw_info.fw_ver[1]),
                     false, "N/A").c_str());
 
@@ -1856,13 +1856,13 @@ bool BurnSubCommand::checkFwVersion(bool CreateFromImgInfo, u_int16_t fw_ver0, u
 
     printf("    New FW version:               ");
     if (CreateFromImgInfo) {
-        snprintf(new_ver, 124,
+        snprintf(new_ver, 124, "%s",
                 new_version.get_fw_version(
                         VERSION_FORMAT(_imgInfo.fw_info.fw_ver[1]), false,
                         "N/A").c_str());
     }
     else {
-        snprintf(new_ver, 124,
+        snprintf(new_ver, 124, "%s",
                 new_version.get_fw_version(VERSION_FORMAT(fw_ver1), false,
                         "N/A").c_str());
     }
```

tzafrir-mellanox, 2021-02-04:
Code no longer exists in the code. It was rewritten in the following commit:

commit e6695575ee6ddc9487a3afbd7fc1132e9fb08c63
Author: Dan Goldberg <dang@mellanox.com>
Date:   Wed Jun 17 23:54:08 2020 +0300

    Fixed format-security warning in flint's checFwVersion
    
    Description:
    Removed redundant snprintf into a char array.
    
    Issue: 2170787

Please close this issue.

Honggang-LI, 2021-02-04:
Thanks for fix this issue.

closed 2021-02-04 by Honggang-LI