Mellanox/mstflint #414: C++ errors in jsoncpp-headers

UnitedMarsupials, 2021-09-01:
Trying to build the tool on FreeBSD (both 11.3 and 12.2) with jsoncpp-1.9.4 installed, I get these compilation errors:
```
mlxarchive_mfa2_builder.cpp:124:38: error: no viable conversion from 'Json::Value::const_iterator' (aka 'Json::ValueConstIterator') to 'Json::ValueIterator'
            for (Json::ValueIterator itr = _componentsJSON.begin();
                                     ^     ~~~~~~~~~~~~~~~~~~~~~~~
/opt/include/json/value.h:884:3: note: candidate constructor not viable: no known conversion from 'Json::Value::const_iterator' (aka 'Json::ValueConstIterator') to 'const Json::ValueIterator &' for 1st argument
  ValueIterator(const ValueIterator& other);
  ^
/opt/include/json/value.h:883:12: note: explicit constructor is not a candidate
  explicit ValueIterator(const ValueConstIterator& other);
           ^
/opt/include/json/value.h:889:12: note: explicit constructor is not a candidate
  explicit ValueIterator(const Value::ObjectValues::iterator& current);
           ^
mlxarchive_mfa2_builder.cpp:159:30: error: no viable conversion from 'Json::Value::const_iterator' (aka 'Json::ValueConstIterator') to 'Json::ValueIterator'
    for (Json::ValueIterator itr = _componentsJSON.begin();
                             ^     ~~~~~~~~~~~~~~~~~~~~~~~
/opt/include/json/value.h:884:3: note: candidate constructor not viable: no known conversion from 'Json::Value::const_iterator' (aka 'Json::ValueConstIterator') to 'const Json::ValueIterator &' for 1st argument
  ValueIterator(const ValueIterator& other);
  ^
/opt/include/json/value.h:883:12: note: explicit constructor is not a candidate
  explicit ValueIterator(const ValueConstIterator& other);
           ^
/opt/include/json/value.h:889:12: note: explicit constructor is not a candidate
  explicit ValueIterator(const Value::ObjectValues::iterator& current);
           ^
3 warnings and 2 errors generated.
```

UnitedMarsupials, 2021-09-01:
Seems like a simple fix:
```diff
--- mlxarchive/mlxarchive_mfa2_builder.cpp    2021-01-26 10:39:55.000000000 -0500
+++ mlxarchive/mlxarchive_mfa2_builder.cpp    2021-09-01 10:46:44.356557000 -0400
@@ -122,5 +122,5 @@
             string name = _deviceDescriptorsJSON[i][COMPONENTS_STR][k].asString();
             int j = 0;
-            for (Json::ValueIterator itr = _componentsJSON.begin();
+            for (auto itr = _componentsJSON.begin();
                             itr != _componentsJSON.end(); itr++, j++) {
                 Json::Value componentJSON = _componentsJSON[name];
@@ -157,5 +157,5 @@
     vector<Component> components;
 
-    for (Json::ValueIterator itr = _componentsJSON.begin();
+    for (auto itr = _componentsJSON.begin();
                 itr != _componentsJSON.end(); itr++) {
         string name = itr.key().asString();

```

ahmadawwad, 2022-01-13:
this code is not existed any more

closed 2022-01-13 by ahmadawwad