Mellanox/mstflint #513: failed to build mstflint with gcc-12

Honggang-LI, 2022-01-26:
```
https://koji.fedoraproject.org/koji/taskinfo?taskID=81799494
https://kojipkgs.fedoraproject.org//work/tasks/9494/81799494/build.log
```
```
make[3]: Leaving directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/iniParser'
Making all in json
make[3]: Entering directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
/bin/sh ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../..  -W -Werror -g -MP -MD -fPIC   -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c -o json_value.lo json_value.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -W -Werror -g -MP -MD -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c json_value.cpp  -fPIC -DPIC -o .libs/json_value.o
make[3]: Leaving directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
json_value.cpp: In member function 'std::string Json::Value::asString() const':
json_value.cpp:23:51: error: this statement may fall through [-Werror=implicit-fallthrough=]
   23 | #define JSON_ASSERT_MESSAGE( condition, message ) if (!( condition )) throw std::runtime_error( message );
      |                                                   ^~
json_value.cpp:706:7: note: in expansion of macro 'JSON_ASSERT_MESSAGE'
  706 |       JSON_ASSERT_MESSAGE( false, "Type is not convertible to string" );
      |       ^~~~~~~~~~~~~~~~~~~
json_value.cpp:707:4: note: here
  707 |    default:
      |    ^~~~~~~
json_value.cpp: In member function 'Json::Value::Int Json::Value::asInt() const':
json_value.cpp:23:51: error: this statement may fall through [-Werror=implicit-fallthrough=]
   23 | #define JSON_ASSERT_MESSAGE( condition, message ) if (!( condition )) throw std::runtime_error( message );
      |                                                   ^~
json_value.cpp:741:7: note: in expansion of macro 'JSON_ASSERT_MESSAGE'
  741 |       JSON_ASSERT_MESSAGE( false, "Type is not convertible to int" );
      |       ^~~~~~~~~~~~~~~~~~~
json_value.cpp:742:4: note: here
  742 |    default:
      |    ^~~~~~~
json_value.cpp: In member function 'Json::Value::UInt Json::Value::asUInt() const':
json_value.cpp:23:51: error: this statement may fall through [-Werror=implicit-fallthrough=]
   23 | #define JSON_ASSERT_MESSAGE( condition, message ) if (!( condition )) throw std::runtime_error( message );
      |                                                   ^~
json_value.cpp:768:7: note: in expansion of macro 'JSON_ASSERT_MESSAGE'
  768 |       JSON_ASSERT_MESSAGE( false, "Type is not convertible to uint" );
      |       ^~~~~~~~~~~~~~~~~~~
json_value.cpp:769:4: note: here
  769 |    default:
      |    ^~~~~~~
json_value.cpp: In member function 'double Json::Value::asDouble() const':
json_value.cpp:23:51: error: this statement may fall through [-Werror=implicit-fallthrough=]
   23 | #define JSON_ASSERT_MESSAGE( condition, message ) if (!( condition )) throw std::runtime_error( message );
      |                                                   ^~
json_value.cpp:793:7: note: in expansion of macro 'JSON_ASSERT_MESSAGE'
  793 |       JSON_ASSERT_MESSAGE( false, "Type is not convertible to double" );
      |       ^~~~~~~~~~~~~~~~~~~
json_value.cpp:794:4: note: here
  794 |    default:
      |    ^~~~~~~
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:470: json_value.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Entering directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
/bin/sh ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../..  -W -Werror -g -MP -MD -fPIC   -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c -o json_reader.lo json_reader.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -W -Werror -g -MP -MD -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c json_reader.cpp  -fPIC -DPIC -o .libs/json_reader.o
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -W -Werror -g -MP -MD -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c json_reader.cpp -o json_reader.o >/dev/null 2>&1
make[3]: Leaving directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
make[3]: Entering directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
/bin/sh ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../..  -W -Werror -g -MP -MD -fPIC   -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c -o json_writer.lo json_writer.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -W -Werror -g -MP -MD -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c json_writer.cpp  -fPIC -DPIC -o .libs/json_writer.o
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -W -Werror -g -MP -MD -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DMST_UL -c json_writer.cpp -o json_writer.o >/dev/null 2>&1
make[3]: Leaving directory '/builddir/build/BUILD/mstflint-4.18.0/ext_libs/json'
make[2]: *** [Makefile:422: all-recursive] Error 1
make[1]: *** [Makefile:551: all-recursive] Error 1
make: *** [Makefile:438: all] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.oReImc (%build)
RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.oReImc (%build)
```

Honggang-LI, 2022-01-27:
https://bugzilla.redhat.com/show_bug.cgi?id=2046760

tomer540, 2022-01-30:
The fedora36 OS is not yet supported.

closed 2022-01-30 by tomer540