Mellanox/mstflint #550: new Coverity issues in 4.19 from 4.18

michich, 2022-03-29:
Coverity scan found several new issues in mstflint 4.19 compared to 4.18. 
Attached is the output report. At least some of the findings look like real bugs. Please check.
[added.err.txt](https://github.com/Mellanox/mstflint/files/8372711/added.err.txt)

tomer540, 2022-03-31:
thank you for the report, i have passed it to the relevant people.
I will update this thread once the issue is clarified.

michich, 2022-08-04:
Here are issues reported by covscan static checkers (clang, coverity, cppcheck, gcc, shellcheck, unicontrol) static checkers in mstflint 4.21.0:
[scan-results.err.txt](https://github.com/Mellanox/mstflint/files/9257564/scan-results.err.txt)

closed 2024-08-05 by github-actions[bot]