Mellanox/mstflint #590: mstconfig not working w/ ConnectX-3

kfandre, 2022-05-28:
This is the same issue as was previously submitted but closed here: [https://github.com/Mellanox/mstflint/issues/33](https://github.com/Mellanox/mstflint/issues/33)

I am unable to enable SR-IOV on this card because the tool can't query or change the config.

I have the same card in a windows machine and I actually have the same problem using the latest WinMFT install from nvidia/mellanox site.

I also tried using the mellanox OFED drivers installed on a supported linux distro and I get the same problem.

Since the connectX-3 isnt supported anymore, this tool is basically the only way to modify the firmware configuration of this otherwise excellent card.  It would be great if someone could figure this out.
 

```
mstflint -d 04:00.0 q
Image type:            FS2
FW Version:            2.42.5000
FW Release Date:       5.9.2017
Product Version:       02.42.50.00
Rom Info:              type=PXE version=3.4.752
Device ID:             4099
Description:           Node             Port1            Port2            Sys image
GUIDs:                 0002c90300056aa8 0002c90300056aa9 0002c90300056aaa 0002c90300056aab
MACs:                                       0002c9a24ac0     0002c9a24ac1
VSD:
PSID:                  MT_1170110023

---

mstconfig -d 04:00.0 q

Device #1:
----------

Device type:    ConnectX3
Device:         04:00.0

Configurations:                              Next Boot
-E- Failed to query device current configuration

---

 mlxconfig -d /dev/mst/mt4099_pciconf0 set SRIOV_EN=1 NUM_OF_VFS=8

Device #1:
----------

Device type:    ConnectX3
Device:         /dev/mst/mt4099_pciconf0

Configurations:                              Next Boot       New
-E- Failed to query device current configuration

```

stapelberg, 2022-09-19:
Running into the same issue. Tried configuring two different ConnectX-3 cards in two different machines, both give the same error message.

stapelberg, 2022-09-20:
Thanks for the tip! Unfortunately, I’m not interested in SR-IOV, I’m interested in Wake On LAN (WOL).

While I can run `flint -d /dev/mst/mt4099_pciconf0 dc`, I can’t find any setting related to WOL in the resulting .ini file.

kfandre, 2022-09-20:
> I was never able to get `mstconfig` working for this, but I was still able to modify the CX3 configs using a combination of `flint` to dump the config file and `mlxburn` to apply it after I made my changes.
> 
> This Reddit post that has a link to the `.mlx` firmware needed (instead of the `.bin` versions available directly from Mellanox's website). https://www.reddit.com/r/homelab/comments/jloll6/comment/gbff04p/?utm_source=share&utm_medium=web2x&context=3
> 
> This STH post outlines how to modify the config. https://forums.servethehome.com/index.php?threads/sr-iov-for-mellanox-connectx-2.12693/#post-121167
> 
> In order to run the `mlxburn` command I needed the `mft-oem` version of the `mft` package which contains the `mic` tool. You should be able to find the version for your machine somewhere here: https://linux.mellanox.com/public/repo/
> 
> Hope this helps.

Ran across this too but it didn't work for me either.  I do recall having a difficult time locating a working mic program. I gave up and bought an x510-da2 based card that just works without all the fuss.

kevindd992002, 2023-12-26:
I just found why this is happening. It seems to be an issues with the mft tools and the single port CX3 (as discussed [here](https://forums.servethehome.com/index.php?threads/connectx-3-custom-firmware-files.39457/)). The solution is to generate an image with a modified ini (to enable sriov) but the mlx fw file you can use with that is older (2.40.5030) than the latest available bin (2.42.5000). I did just that and it worked. I can see the vf's in proxmox now.

However, is it worth running an older fw to make sriov work or maybe just replace this card with a CX4 or even a dual port CX3 that don't exhibit this issue? With these cards, you can simply use mlxconfig to edit the fw settings directly without needing to generate a modified image and reflashing.

So does anyone have a link to the 2.42.5000 mlx fw?

EasyNetDev, 2023-12-30:
Hi,

I'm facing the same issue with version 4.26.0. On 4.25.0 is working without issues. But the strange thing is that in PDF is still says on page 10 and 11:
Supported Interface Cards (NICs): Group I/4th Generation -> Adapter Cards: NVIDIA ConnectX-3, NVIDIA ConnectX-3 Pro.

Even with the driver is still says:
```
mstconfig query
-E- Unsupported device
-E- Unsupported device
```

With 4.25.0:
```
mstconfig query | grep Device
Device #1:
Device type:    ConnectX3
Device:         /sys/bus/pci/devices/0000:03:00.0/config
Device #2:
Device type:    ConnectX3
Device:         /sys/bus/pci/devices/0000:04:00.0/config
```

I'm using Debian Trixie:
```
Linux R02 6.5.0-4-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.10-1 (2023-11-03) x86_64 GNU/Linux
```

J4nsen, 2024-02-08:
Same for me with a Connect X3 Pro. Version 4.26 reports `-E- Unsupported device`, but version 4.25 works.

tk154, 2024-07-23:
The `-E- Unsupported device` error for 4th gen cards was an intended change made by 52eb6c0ebbe8ad585da2a00c751260a025ad3f20.

ogalbxela, 2024-08-06:
4th generation adapters support was removed starting 4.26
Release notes for 4.29 will be adjusted to reflect that..

closed 2024-08-06 by ogalbxela

adavidzh, 2025-09-01:
> Same for me with a Connect X3 Pro. Version 4.26 reports `-E- Unsupported device`, but version 4.25 works.

This was a life saver. FWIW, 4.25.1 for Linux also works.