Mellanox/mstflint #712: Configuration does not check for the presence of infiniband libs/headers

blochl, 2023-01-11:
The `./configure` stage passes with no errors, but the build fails for missing `infiniband/verbs.h`, if rdma-core is not installed.

ogalbxela, 2023-01-13:
Hi blochl,

We are aware about the issue and working on getting it resolved.
Would you mind sharing the build log, info about Linux distribution you are using as well as the version of Linux kernel?
What mstflint branch you are using to build?

Meanwhile, your options are:
  - you may mitigate the issue by installing libibverbs-dev.
  - you may find all mstflint dependencies in the https://github.com/Mellanox/mstflint/blob/master_devel/debian/control
  - you may use dpkg-buildpackage way of building - it will explicitly show you if any dependencies are missed met on you machine before building:
            > dpkg-buildpackage -uc -us
            > dpkg -i ../mstflint_<version>_<arch>.deb
            > dpkg -i ../mstflint-dkms_<version>_all.deb

blochl, 2023-01-13:
The proposed fix is in https://github.com/Mellanox/mstflint/pull/713.

closed 2023-09-04 by ahmadawwad