Mellanox/mstflint #734: Register values set with `mlxreg` are not persistent across reboots

kcgthb, 2023-02-03:
When setting a register on a switch with `mlxreg`, the written values are not persistent and are lost after a reboot (the register    value is reset to its default value).

Here's an example trying to modify the `node_description[0]` field in the `SPZR` register for a MQM8790 switch (fw 27.2010.3118) using MFT 4.23:

1. checking the original value:
   ```
   # mlxreg -d lid-801 --reg_name SPZR --get --indexes router_entity=0x0,swid=0x0 | grep "node_description\[0\]"
   node_description[0] | 0x5175616e
   ```

2. trying to set a new value to `0x00000000`:
   ```
   # mlxreg -d lid-801 --reg_name SPZR --set 'ndm=0x1,node_description[0]=0x00000000' --indexes router_entity=0x0,swid=0x0 --yes
   You are about to send access register: SPZR with the following data:
   Field Name              | Data
   =====================================
   enh_sw_p0               | 0x00000000
   g0                      | 0x00000000
   ng                      | 0x00000000
   sig                     | 0x00000000
   mp                      | 0x00000000
   vk                      | 0x00000000
   cm                      | 0x00000000
   enh_sw_p0_mask          | 0x00000000
   ndm                     | 0x00000001
   cm2                     | 0x00000000
   router_entity           | 0x00000000
   swid                    | 0x00000000
   capability_mask         | 0x4450d848
   system_image_guid_h     | 0xb83fd203
   system_image_guid_l     | 0x00d7f97a
   guid0_h                 | 0xb83fd203
   guid0_l                 | 0x00d7f97a
   node_guid_h             | 0xb83fd203
   node_guid_l             | 0x00d7f97a
   capability_mask2        | 0x0000003b
   max_pkey                | 0x00000008
   node_description[0]     | 0x00000000
   node_description[1]     | 0x74756d20
   node_description[2]     | 0x4d656c6c
   node_description[3]     | 0x616e6f78
   node_description[4]     | 0x20546563
   node_description[5]     | 0x686e6f6c
   node_description[6]     | 0x6f676965
   node_description[7]     | 0x73000000
   node_description[8]     | 0x00000000
   node_description[9]     | 0x00000000
   node_description[10]    | 0x00000000
   node_description[11]    | 0x00000000
   node_description[12]    | 0x00000000
   node_description[13]    | 0x00000000
   node_description[14]    | 0x00000000
   node_description[15]    | 0x00000000
   =====================================
   
    Do you want to continue ? (y/n) [n] : y
    Sending access register...
   # echo $?
   0
   ```

3. checking that the new value has correctly been set:
   ```
   # mlxreg -d lid-801 --reg_name SPZR --get --indexes router_entity=0x0,swid=0x0 | grep "node_description\[0\]"
   node_description[0]     | 0x00000000
   ```

4. rebooting the switch
   ```
   # flint -d lid-801 swreset
   -I- Sending reset command to device lid-800 ...
   -I- Reset command accepted by the device.
   ```

5. checking the value again:
   ```
   # mlxreg -d lid-801 --reg_name SPZR --get --indexes router_entity=0x0,swid=0x0 | grep "node_description\[0\]"
   node_description[0] | 0x5175616e
   ```

After the switch reboot, the `node_description[0]` attribute has been reset to the initial, default value (`0x5175616e`) and the value that has been set previously (`0x00000000`) is lost.

Is this an issue with `mlxreg`, or with the switch firmware?

Thanks!

kcgthb, 2024-05-08:
The issue is still pertinent.

ogalbxela, 2024-08-06:
The MSTFLINT set of applets for NIC firmware manipulation does not control the behavior of the NIC after an adapter reboot. That falls under the firmware's responsibilities.

closed 2024-08-06 by ogalbxela

kcgthb, 2024-08-07:
Yet there seems to be ways to set a persistent node description, that persists after a reboot, so the firmware seems to be allowing this already.