Mellanox/mstflint #785: Dependency with pciutils

wizhaoredhat, 2023-05-26:
Without pciutils, we will get the following error message when running mstfwreset:

Continue with reset?[y/N] y
Failed
-E- failed to run 'setpci -s 0000:c9:02.0 0x0.w'.

The setpci command is part of the pciutils package.

@bn222 @SalDaniele

bn222, 2023-05-30:
afair, pciutils should be part of the container. I've purposefully checked that. If not, it got removed which is something we need to catch from CI.

SalDaniele, 2023-05-30:
We noticed the error due to failing CI jobs after the most recent firmware update. Installing pci utils in the config-daemon image results in mstfwreset running successfully https://github.com/k8snetworkplumbingwg/sriov-network-operator/pull/449

wizhaoredhat, 2023-05-30:
@adrianchiris

closed 2024-11-01 by github-actions[bot]