Mellanox/mstflint #797: v4.24.0 not "fully" released

nmorey, 2023-07-03:
@ogalbxela 

https://github.com/Mellanox/mstflint/releases/tag/v4.24.0-1 is missing some assets compare to the older releases:
- The userland and kernel release tarball (just the source ones are available)
- No release notes/doc

As the maintainer of this package for SUSE, just wondering if this is an oversight? Or a new workflow?

ogalbxela, 2023-07-04:
Thank you for your attention to this matter.
There haven't been any changes in the flow. Both release user and kernel space tarballs were uploaded at the time the release was issued, and they have been available on the release page ever since. 
I have copied them under the "assets"-section for your convenience.
The release notes were delayed, but I have uploaded them today.
Please let us know if that resolves your concern.

nmorey, 2023-07-17:
All good thanks :)

closed 2023-08-08 by ogalbxela