Mellanox/mstflint #830: mstfwreset fails when card is in DPU mode

SalDaniele, 2023-08-23:
Using mstflint compiled from source code:

```
# mstflint -v
mstflint, mstflint 4.25.0, built on Aug 23 2023, 19:39:15. Git SHA Hash: N/A
```

On a Bluefield-2 w/ BMC
```
Device #1:
----------

Device type:    BlueField2      
Name:           MBF2H512C-AECO_Ax
Description:    BlueField-2 P-Series DPU 25GbE Dual-Port SFP56; integrated BMC; PCIe Gen4 x8; Secure Boot Enabled; Crypto Enabled; 16GB on-board DDR; 1GbE OOB management; FHHL
Device:         ca:00.0         

Configurations:                                      Default         Current         Next Boot
         MEMIC_BAR_SIZE                              0               0               0               
         MEMIC_SIZE_LIMIT                            _256KB(1)       _256KB(1)       _256KB(1)       
         HOST_CHAINING_MODE                          DISABLED(0)     DISABLED(0)     DISABLED(0)     
         HOST_CHAINING_CACHE_DISABLE                 False(0)        False(0)        False(0)        
         HOST_CHAINING_DESCRIPTORS                   Array[0..7]     Array[0..7]     Array[0..7]     
         HOST_CHAINING_TOTAL_BUFFER_SIZE             Array[0..7]     Array[0..7]     Array[0..7]     
         INTERNAL_CPU_MODEL                          EMBEDDED_CPU(1) EMBEDDED_CPU(1) EMBEDDED_CPU(1) 
         INTERNAL_CPU_PAGE_SUPPLIER                  ECPF(0)         ECPF(0)         ECPF(0)         
         INTERNAL_CPU_ESWITCH_MANAGER                ECPF(0)         ECPF(0)         ECPF(0)         
         INTERNAL_CPU_IB_VPORT0                      ECPF(0)         ECPF(0)         ECPF(0)         
         INTERNAL_CPU_OFFLOAD_ENGINE                 ENABLED(0)      ENABLED(0)      ENABLED(0)  
         FLEX_PARSER_PROFILE_ENABLE                  0               0               0               
         PROG_PARSE_GRAPH                            False(0)        False(0)        False(0)        
         FLEX_IPV4_OVER_VXLAN_PORT                   0               0               0               
         ROCE_NEXT_PROTOCOL                          254             254             254             
         ESWITCH_HAIRPIN_DESCRIPTORS                 Array[0..7]     Array[0..7]     Array[0..7]     
         ESWITCH_HAIRPIN_TOT_BUFFER_SIZE             Array[0..7]     Array[0..7]     Array[0..7]     
         PF_BAR2_SIZE                                3               3               3               
         DPU_RESET_NOTIFICATION_ENABLED              ENABLED(1)      ENABLED(1)      ENABLED(1)      
         INTERNAL_CPU_RSHIM                          ENABLED(0)      ENABLED(0)      ENABLED(0)      
         PF_NUM_OF_VF_VALID                          False(0)        False(0)        False(0)        
         NON_PREFETCHABLE_PF_BAR                     False(0)        False(0)        False(0)        
         VF_VPD_ENABLE                               False(0)        False(0)        False(0)        
         PF_NUM_PF_MSIX_VALID                        False(0)        False(0)        False(0)        
         PER_PF_NUM_SF                               False(0)        False(0)        False(0)        
         STRICT_VF_MSIX_NUM                          False(0)        False(0)        False(0)        
         VF_NODNIC_ENABLE                            False(0)        False(0)        False(0)        
         NUM_PF_MSIX_VALID                           True(1)         True(1)         True(1)         
         NUM_OF_VFS                                  8               8               8               
         NUM_OF_PF                                   2               2               2               
         PF_BAR2_ENABLE                              True(1)         True(1)         True(1)         
         HIDE_PORT2_PF                               False(0)        False(0)        False(0)        
         SRIOV_EN                                    True(1)         True(1)         True(1)         
         PF_LOG_BAR_SIZE                             5               5               5               
         VF_LOG_BAR_SIZE                             0               0               0               
         NUM_PF_MSIX                                 63              63              63              
         NUM_VF_MSIX                                 11              11              11              
         INT_LOG_MAX_PAYLOAD_SIZE                    AUTOMATIC(0)    AUTOMATIC(0)    AUTOMATIC(0)    
         PCIE_CREDIT_TOKEN_TIMEOUT                   0               0               0               
         RT_PPS_ENABLED_ON_POWERUP                   False(0)        False(0)        False(0)        
         LAG_RESOURCE_ALLOCATION                     DEVICE_DEFAULT(0) DEVICE_DEFAULT(0) DEVICE_DEFAULT(0)
         PHY_COUNT_LINK_UP_DELAY                     DELAY_NONE(0)   DELAY_NONE(0)   DELAY_NONE(0)   
         ACCURATE_TX_SCHEDULER                       False(0)        False(0)        False(0)        
         PARTIAL_RESET_EN                            False(0)        False(0)        False(0)        
         RESET_WITH_HOST_ON_ERRORS                   False(0)        False(0)        False(0)        
         NVME_EMULATION_ENABLE                       False(0)        False(0)        False(0)        
         NVME_EMULATION_NUM_VF                       0               0               0               
         NVME_EMULATION_NUM_PF                       1               1               1               
         NVME_EMULATION_VENDOR_ID                    5555            5555            5555            
         NVME_EMULATION_DEVICE_ID                    24577           24577           24577           
         NVME_EMULATION_CLASS_CODE                   67586           67586           67586           
         NVME_EMULATION_REVISION_ID                  0               0               0               
[... 285 lines cut ...]
Continue with reset?[y/N] y
-I- Sending Reset Command To Fw             -Done
-I- Stopping Driver                         -Done
-I- Resetting PCI                           -Done
-I- Starting Driver                         -Done
-I- FW was loaded successfully.

```
# mstflint -d ca:00.0 q
Image type:            FS4
FW Version:            24.38.1002
FW Release Date:       3.8.2023
Product Version:       24.38.1002
Rom Info:              type=UEFI Virtio net version=21.4.10 cpu=AMD64,AARCH64
                       type=UEFI Virtio blk version=22.4.10 cpu=AMD64,AARCH64
                       type=UEFI version=14.31.20 cpu=AMD64,AARCH64
                       type=PXE version=3.7.201 cpu=AMD64
Description:           UID                GuidsNumber
Base GUID:             e8ebd30300cee2e6        16
Base MAC:              e8ebd3cee2e6            16
Image VSD:             N/A
Device VSD:            N/A
PSID:                  MT_0000000724
Security Attributes:   secure-fw
```

thillux, 2023-11-05:
We also experienced such behavior with the current master_devel branch of the upcoming 4.26 release. Do you plan to fix this with 4.26?

We use mstflint together with a mainline kernel 6.5 at the moment.

thillux, 2023-11-06:
Update: with kernel 6.6 and a more recent rdma-core version we were able to trigger a reset succesfully from BF2. But ran into the 60s timeout. dmesg showed, that the reset worked.

ogalbxela, 2024-05-08:
Had a conversation with the owner from our side.
Direction was: please use mstflint-4.28 (just released) and the latest available driver to flash the latest published firmware.
Please also query the device with "mstfwreset" (mstfwreset -d DEVICE q). It will list "sync"-capabilities for you.

Something like:
```shell
mstfwreset -d 81:00.0 q
 
<some output omitted>

Reset-sync (relevant only for reset-level 3):
0: Tool is the owner                                             -Not supported 
1: Driver is the owner                                           -Supported     (default)
```

For "sync 0" - tool is the owner of reset flow and **reset command should be issued from both host and arm side**
For "sync 1" - driver is the owner

abhiramnarayana, 2024-11-04:
Hi,

Even we faced the same issue of "Synchronization by driver is not supported in the current state of this device.".
Device we are using is as below
~]$ sudo lshw -class network -businfo | grep BlueField-2
pci@0000:17:00.0  ens2f0np0  network        MT42822 BlueField-2 integrated ConnectX-6 Dx network controller
pci@0000:17:00.1  ens2f1np1  network        MT42822 BlueField-2 integrated ConnectX-6 Dx network controller

As suggested in the thread above we updated the Firmware to latest (24.42.1000) and also took v4.29 mstflint latest. But still we are facing issue during mstfwreset. (Even with v4.28 it gives the same error as in v4.29)
 ~]$ ethtool -i ens2f0np0
driver: mlx5_core
version: 5.14.0-427.42.1.el9_4.x86_64
firmware-version: 24.42.1000 (MT_0000000765)
expansion-rom-version: 
bus-info: 0000:17:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes
[cloud-admin@compute-1 ~]$

~]$ sudo mstflint --version
mstflint, mstflint 4.29.0, Git SHA Hash: 37af981

~]$ sudo mstfwreset -d 17:00.0 q

Reset-levels:
0: Driver, PCI link, network link will remain up ("live-Patch")  -Not Supported 
1: Only ARM side will not remain up ("Immediate reset").         -Not Supported 
3: Driver restart and PCI reset                                  -Supported     (default)
4: Warm Reboot                                                   -Supported     

Reset-types (relevant only for reset-levels 1,3,4):
0: Full chip reset                                               -Supported     (default)
1: Phy-less reset (keep network port active during reset)        -Not Supported 
2: NIC only reset (for SoC devices)                              -Not Supported 
3: ARM only reset                                                -Not Supported 
4: ARM OS shut down                                              -Not Supported 

Reset-sync (relevant only for reset-level 3):
0: Tool is the owner                                             -Not supported 
1: Driver is the owner                                           -Supported     (default)

Reset-reason:                                                    Warm reset    
Timestamp (number of clock cycles) since last cold reset:        1308350112    

Note that the Reset-sync (sync 1) shows supported. But still it is not working.
 ~]$ sudo mstfwreset --device 0000:17:00.0 --level 3 -y r

The reset level for device, 0000:17:00.0 is:

3: Driver restart and PCI reset
Please be aware that resetting the Bluefield may take several minutes. Exiting the process in the middle of the waiting period will not halt the reset.
The ARM side will be restarted, and it will be unavailable for a while.
Continue with reset?[y/N] y
**-I- Sending Reset Command To Fw             --E- The BF reset flow encountered a failure due to a reset state error of negotiation dis-acknowledgment.**
[cloud-admin@compute-1 ~]$

closed 2026-04-09 by github-actions[bot]