Mellanox/mstflint #835: mstflint segmentation fault when querying switch in-band

NateCrawford, 2023-09-15:
I'm getting segmentation faults with mstflint 4.20.0 and above when querying a switch. v4.18.0 seems to work fine.

4.18.0:
```
mstflint -d lid-9 q
Image type:            FS3
FW Version:            15.1100.0072
FW Release Date:       9.6.2016
Description:           UID                GuidsNumber
Base GUID:             248a070300fab5b0        32
Base MAC:              248a07fab5b0            32
Image VSD:             N/A
Device VSD:            N/A
PSID:                  MT_2640110032
Security Attributes:   N/A
```
4.20.0:
```
mstflint -d lid-9 q
Segmentation fault (core dumped)
```

Running Rocky 8.8 with 4.18.0-477.21.1.el8_8.x86_64 rdma drivers. Using default compiler (GCC 8.5.0) to build mstflint. 
The package from the Appstream repository (mstflint-4.23.0-2.el8.x86_64) also segfaults.

Querying the HCA works fine with all versions.

Did something significant change in 4.20?

ogalbxela, 2024-07-18:
please post the model of device in question

knweiss, 2024-10-22:
FWIW I can reproduce this issue with the official `mstflint` binaries of Rocky Linux 8.10 and 9.4 with MQM8790-HS2X_Ax or SSG7A41878_SSG7A41881_Ax (Mellanox QM8790) unmanaged IB switches.

- mstflint-4.25.0-1.el8.x86_64 (segfaults)
- mstflint-4.25.0-1.el9.x86_64 (segfaults)

Workarounds that seem to work from my limited testing:
- **Downgrade** to the mstflint binary from mstflint-4.18.0-1.el8.x86_64 of rhel86 for rocky810 or https://oraclelinux.pkgs.org/9/ol9-appstream-x86_64/mstflint-4.18.0-1.el9.x86_64.rpm.html for rocky94.
- **Upgrade** to `flint` from at least mft-4.26.1 from NVIDIA (https://network.nvidia.com/products/adapter-software/firmware-tools/).