Mellanox/mstflint #861: --enable-dpa does no compile

thillux, 2023-10-31:
```
/nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: ../mlxfwops/lib/libmlxfwops.a(fs_cert_ops.o): in function `FsCertOperations::FwInit()':
/build/source/mlxfwops/lib/components/fs_cert_ops.cpp:47: undefined reference to `CertStructHeader::CertStructHeader()'
/nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: /build/source/mlxfwops/lib/components/fs_cert_ops.cpp:54: undefined reference to `CertStructHeader::Deserialize(std::vector<unsigned char, std::allocator<unsigned char> >)'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:624: mstfwmanager] Error 1
make[2]: Leaving directory '/build/source/mlxfwupdate'
make[1]: *** [Makefile:555: all-recursive] Error 1
make[1]: Leaving directory '/build/source'
make: *** [Makefile:440: all] Error 2
error: build of '/nix/store/r7r209cll964zpbhx17kx3qfl8rvn0ak-mstflint-4.26.0-1.drv' on 'ssh-ng://root@nixos-dev-01' failed: builder for '/nix/store/r7r209cll964zpbhx17kx3qfl8rvn0ak-mstflint-4.26.0-1.drv' failed with exit code 2;
       last 10 log lines:
       > libtool: link: g++ -g -MP -MD -Wall -W -DMSTFLINT -I../ext_libs/iniParser -I../ext_libs/json -DUSE_XML -DLIBXML_STATIC -DUSE_CURL -DCURL_STATICLIB -g -O2 -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DNO_RDMEM -DMST_UL -o mstfwmanager mstfwmanager-cmd_line_params.o mstfwmanager-cmd_line_parser.o mstfwmanager-fw_version_old_clp.o mstfwmanager-fw_version_with_sub_build.o mstfwmanager-image_access.o mstfwmanager-img_version.o mstfwmanager-menu.o mstfwmanager-mlnx_dev.o mstfwmanager-mlxfwmanager.o mstfwmanager-mlxfwmanager_common.o mstfwmanager-output_fmts.o mstfwmanager-psid_lookup_db.o mstfwmanager-psid_query_item.o mstfwmanager-server_request.o  -L/usr/local/lib -L/usr/local/lib64 -L/usr/lib64 ../dev_mgt/.libs/libdev_mgt.a ../mvpd/.libs/libmvpd.a ../mlxfwops/lib/libmlxfwops.a ../pldmlib/.libs/libpldm.a ../mlxconfig/.libs/libmlxcfg.a /build/source/mtcr_ul/.libs/libmtcr_ul.a ../libmfa/.libs/libmfa.a ../ext_libs/minixz/.libs/libminixz.a ../cmdparser/libcmdparser.a ../mflash/.libs/libmflash.a ../tools_res_mgmt/libtools_res_mgmt.a ../mft_utils/libmftutils.a ../cmdif/.libs/libcmdif.a ../reg_access/.libs/libreg_access.a ../tools_layouts/.libs/libtools_layouts.a ../fw_comps_mgr/libfw_comps_mgr.a ../mtcr_ul/.libs/libmtcr_ul.a ../ext_libs/json/.libs/libjson.a /nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64/libstdc++.so ../ext_libs/iniParser/.libs/libiniparser.a ../ext_libs/muparser/libmuparser.a ../ext_libs/sqlite/libsqlite3.a ../mlxsign_lib/libmlxsign.a ../tools_crypto/libtools_crypto.a ../mad_ifc/libmad_ifc.a -lm -ldl /nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib/libcurl.so -lrt -lssl -lcrypto /nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib/libxml2.so -lz -pthread -Wl,-rpath -Wl,/nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64 -Wl,-rpath -Wl,/nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib -Wl,-rpath -Wl,/nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib -Wl,-rpath -Wl,/nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64 -Wl,-rpath -Wl,/nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib -Wl,-rpath -Wl,/nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib
       > /nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: ../mlxfwops/lib/libmlxfwops.a(fs_cert_ops.o): in function `FsCertOperations::FwInit()':
       > /build/source/mlxfwops/lib/components/fs_cert_ops.cpp:47: undefined reference to `CertStructHeader::CertStructHeader()'
       > /nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: /build/source/mlxfwops/lib/components/fs_cert_ops.cpp:54: undefined reference to `CertStructHeader::Deserialize(std::vector<unsigned char, std::allocator<unsigned char> >)'
       > collect2: error: ld returned 1 exit status
       > make[2]: *** [Makefile:624: mstfwmanager] Error 1
       > make[2]: Leaving directory '/build/source/mlxfwupdate'
       > make[1]: *** [Makefile:555: all-recursive] Error 1
       > make[1]: Leaving directory '/build/source'
       > make: *** [Makefile:440: all] Error 2
       For full logs, run 'nix log /nix/store/r7r209cll964zpbhx17kx3qfl8rvn0ak-mstflint-4.26.0-1.drv'.
error: builder for '/nix/store/r7r209cll964zpbhx17kx3qfl8rvn0ak-mstflint-4.26.0-1.drv' failed with exit code 1

```

ogalbxela, 2023-11-01:
nm

thillux, 2023-11-01:
```
libtool: link: ( cd ".libs" && rm -f "libmstarchive.la" && ln -s "../libmstarchive.la" "libmstarchive.la" )
/nix/store/lf0wpjrj8yx4gsmw2s3xfl58ixmqk8qa-bash-5.2-p15/bin/bash ../libtool  --tag=CXX   --mode=link g++ -Wall -W -g -MP -MD -pipe  -fPIC -g -O2 -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DNO_RDMEM -DMST_UL -shared -L/usr/local/lib -L/usr/local/lib64 -L/usr/lib64 -o mstarchive mlxarchive.o libmstarchive.la ../mlxfwops/lib/libmlxfwops.a ../libmfa/libmfa.la ../fw_comps_mgr/libfw_comps_mgr.la ../mtcr_ul/libmtcr_ul.la ../mflash/libmflash.la ../tools_res_mgmt/libtools_res_mgmt.la ../dev_mgt/libdev_mgt.la ../cmdif/libcmdif.la ../mft_utils/libmftutils.la ../ext_libs/json/libjson.la ../cmdparser/libcmdparser.a ../reg_access/libreg_access.la ../tools_layouts/libmlxarchive_layouts.la ../tools_layouts/libtools_layouts.la ../xz_utils/libxz_utils.la ../ext_libs/minixz/libminixz.la ../xz_utils/libxz_utils.la -lcurl  -lssl -lcrypto -lrt -llzma -lm  -ldl ../mlxsign_lib/libmlxsign.la -lcrypto -lssl ../tools_crypto/libtools_crypto.a -lcrypto ../mad_ifc/libmad_ifc.la -lxml2 -lz -lz 
libtool: link: g++ -Wall -W -g -MP -MD -pipe -fPIC -g -O2 -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DNO_RDMEM -DMST_UL -o .libs/mstarchive mlxarchive.o  -L/usr/local/lib -L/usr/local/lib64 -L/usr/lib64 ./.libs/libmstarchive.so ../mlxfwops/lib/libmlxfwops.a ../libmfa/.libs/libmfa.a ../fw_comps_mgr/.libs/libfw_comps_mgr.a /build/source/mtcr_ul/.libs/libmtcr_ul.so ../mtcr_ul/.libs/libmtcr_ul.so ../mflash/.libs/libmflash.a ../tools_res_mgmt/.libs/libtools_res_mgmt.a ../dev_mgt/.libs/libdev_mgt.a ../cmdif/.libs/libcmdif.a ../mft_utils/.libs/libmftutils.a ../ext_libs/json/.libs/libjson.a ../cmdparser/libcmdparser.a ../reg_access/.libs/libreg_access.a ../tools_layouts/.libs/libmlxarchive_layouts.a ../tools_layouts/.libs/libtools_layouts.a ../ext_libs/minixz/.libs/libminixz.a ../xz_utils/.libs/libxz_utils.a /nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib/libcurl.so -lrt /nix/store/hx3fyk0d9wn7nir8m1qp0y93ixka09lb-xz-5.4.4/lib/liblzma.so -ldl ../mlxsign_lib/.libs/libmlxsign.a -lssl ../tools_crypto/libtools_crypto.a -lcrypto ../mad_ifc/.libs/libmad_ifc.a /nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64/libstdc++.so -lm /nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib/libxml2.so -lz -pthread -Wl,-rpath -Wl,/nix/store/ga4r4xsz2hn6ia24yi68l01q9pxli6a1-mstflint-4.26.0-1/lib -Wl,-rpath -Wl,/nix/store/ga4r4xsz2hn6ia24yi68l01q9pxli6a1-mstflint-4.26.0-1/lib/mstflint -Wl,-rpath -Wl,/nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib -Wl,-rpath -Wl,/nix/store/hx3fyk0d9wn7nir8m1qp0y93ixka09lb-xz-5.4.4/lib -Wl,-rpath -Wl,/nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64 -Wl,-rpath -Wl,/nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib
/nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: ../mlxfwops/lib/libmlxfwops.a(fs_cert_ops.o): in function `FsCertOperations::FwInit()':
/build/source/mlxfwops/lib/components/fs_cert_ops.cpp:47: undefined reference to `CertStructHeader::CertStructHeader()'
/nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: /build/source/mlxfwops/lib/components/fs_cert_ops.cpp:54: undefined reference to `CertStructHeader::Deserialize(std::vector<unsigned char, std::allocator<unsigned char> >)'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:623: mstarchive] Error 1
make[2]: Leaving directory '/build/source/mlxarchive'
make[1]: *** [Makefile:553: all-recursive] Error 1
make[1]: Leaving directory '/build/source'
make: *** [Makefile:438: all] Error 2
error: build of '/nix/store/1z22jxgq56znja5h70yz3jjhc0bbi3zv-mstflint-4.26.0-1.drv' on 'ssh-ng://root@nixos-dev-01' failed: builder for '/nix/store/1z22jxgq56znja5h70yz3jjhc0bbi3zv-mstflint-4.26.0-1.drv' failed with exit code 2;
       last 10 log lines:
       > libtool: link: g++ -Wall -W -g -MP -MD -pipe -fPIC -g -O2 -isystem /usr/local/include -DHAVE_TERMIOS_H -isystem /usr/include/libxml2 -DNO_RDMEM -DMST_UL -o .libs/mstarchive mlxarchive.o  -L/usr/local/lib -L/usr/local/lib64 -L/usr/lib64 ./.libs/libmstarchive.so ../mlxfwops/lib/libmlxfwops.a ../libmfa/.libs/libmfa.a ../fw_comps_mgr/.libs/libfw_comps_mgr.a /build/source/mtcr_ul/.libs/libmtcr_ul.so ../mtcr_ul/.libs/libmtcr_ul.so ../mflash/.libs/libmflash.a ../tools_res_mgmt/.libs/libtools_res_mgmt.a ../dev_mgt/.libs/libdev_mgt.a ../cmdif/.libs/libcmdif.a ../mft_utils/.libs/libmftutils.a ../ext_libs/json/.libs/libjson.a ../cmdparser/libcmdparser.a ../reg_access/.libs/libreg_access.a ../tools_layouts/.libs/libmlxarchive_layouts.a ../tools_layouts/.libs/libtools_layouts.a ../ext_libs/minixz/.libs/libminixz.a ../xz_utils/.libs/libxz_utils.a /nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib/libcurl.so -lrt /nix/store/hx3fyk0d9wn7nir8m1qp0y93ixka09lb-xz-5.4.4/lib/liblzma.so -ldl ../mlxsign_lib/.libs/libmlxsign.a -lssl ../tools_crypto/libtools_crypto.a -lcrypto ../mad_ifc/.libs/libmad_ifc.a /nix/store/9fy9zzhf613xp0c3jsjxbjq6yp8afrsv-gcc-12.3.0-lib/lib/../lib64/libstdc++.so -lm /nix/store/9grbphisvxx14mpbx1n31hg48spca4bx-libxml2-2.11.5/lib/libxml2.so -lz -pthread -Wl,-rpath -Wl,/nix/store/ga4r4xsz2hn6ia24yi68l01q9pxli6a1-mstflint-4.26.0-1/lib -Wl,-rpath -Wl,/nix/store/ga4r4xsz2hn6ia24yi68l01q9pxli6a1-mstflint-4.26.0-1/lib/mstflint -Wl,-rpath -Wl,/nix/store/wxwljhqszqzg2wlvbckc7h61p5k8ap2v-curl-8.4.0/lib -Wl,-rpath -Wl,/nix/store/hx3f [... cut ...]
       > /nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: /build/source/mlxfwops/lib/components/fs_cert_ops.cpp:54: undefined reference to `CertStructHeader::Deserialize(std::vector<unsigned char, std::allocator<unsigned char> >)'
       > collect2: error: ld returned 1 exit status
       > make[2]: *** [Makefile:623: mstarchive] Error 1
       > make[2]: Leaving directory '/build/source/mlxarchive'
       > make[1]: *** [Makefile:553: all-recursive] Error 1
       > make[1]: Leaving directory '/build/source'
       > make: *** [Makefile:438: all] Error 2
       For full logs, run 'nix log /nix/store/1z22jxgq56znja5h70yz3jjhc0bbi3zv-mstflint-4.26.0-1.drv'.
```

thillux, 2023-11-01:
You can find my build flags here: https://github.com/NixOS/nixpkgs/pull/263149

Or more specific configureFlags in: https://github.com/NixOS/nixpkgs/blob/d258451c0ee17626ebcee046a33a4068358bcf57/pkgs/tools/misc/mstflint/default.nix

thillux, 2023-11-01:
As you asked for a full log. Here is one, for the current master_devel branch [build.log](https://github.com/Mellanox/mstflint/files/13224147/build.log)

ogalbxela, 2023-11-01:
> As you asked for a full log. Here is one, for the current master_devel branch [build.log](https://github.com/Mellanox/mstflint/files/13224147/build.log)

Double check darwin. They link it with pre-compiled binary

   ./mlxdpa/dpa_elf/arm64/libdpa_elf
   ./mlxdpa/dpa_elf/x86_64/libdpa_elf
   
   Linux only.

thillux, 2023-11-01:
The nix package is currently marked as linux only. With your recent fixes I was able to build master_devel directly withtout issues! :+1 I'll modify the PR again to only contain the missing fix for recent kernel versions >= 6.4.

When do you currently plan to release 4.26?

closed 2023-11-01 by thillux

ogalbxela, 2023-11-01:
> The nix package is currently marked as linux only. With your recent fixes I was able to build master_devel directly without issues! :+1 I'll modify the PR again to only contain the missing fix for recent kernel versions >= 6.4.
> 
> When do you currently plan to release 4.26?

plan was by the end of October
entire development is in Israel, so may be delayed