Mellanox/mstflint #925: mstconfig cannot access devices with a large PCI domain

acgoldma, 2024-03-29:
```
# lspci | grep Mel
10000:01:00.0 Ethernet controller: Mellanox Technologies MT27800 Family [ConnectX-5]
# mstconfig q
-E- Failed to open the device
# mstconfig -d 10000:01:00.0 q
-E- Failed to open the device
```

Quick check of the code shows that the domain is only stored in a 16-bit value, which it can be larger than. I would suggest expand the fields to 32 bit at least.

ogalbxela, 2024-04-04:
Isn't 10000 fit in 16 bits?
Making the change you suggested in https://github.com/Mellanox/mstflint/pull/926 might hide the actual problem. We'll look into it further internally.

In the meantime, could you please let us know the Linux distribution and version of the lspci utility?

ogalbxela, 2024-04-06:
https://github.com/pciutils/pciutils/commit/ab61451d47514c473953a24aa4f4f816b77ade56

March, 2016 :-)

We'll fix it in MFT then propagate here. Thank you!

ogalbxela, 2025-01-29:
Status here, feature request is initiated internally (#4047673) and is under implemention. Requested support will require more changes than https://github.com/Mellanox/mstflint/pull/926 has