Mellanox/mstflint #928: `mstmcra` cmd cannot read and write the value of the given register

Eternity-Wang, 2024-04-08:
I use the `mstmcra` cmd to read and write the mlx5_1 registers value, but I found that the values of these registers did not change. Can someone help me what the problem is?

`ibv_devinfo -d mlx5_1
hca_id: mlx5_1
        transport:                      InfiniBand (0)
        fw_ver:                         16.32.2004
        node_guid:                      08c0:eb03:00e3:b853
        sys_image_guid:                 08c0:eb03:00e3:b852
        vendor_id:                      0x02c9
        vendor_part_id:                 4121
        hw_ver:                         0x0
        board_id:                       DEL0000000004
        phys_port_cnt:                  1
                port:   1
                        state:                  PORT_ACTIVE (4)
                        max_mtu:                4096 (5)
                        active_mtu:             1024 (3)
                        sm_lid:                 0
                        port_lid:               0
                        port_lmc:               0x00
                        link_layer:             Ethernet

Before:
sudo mstmcra mlx5_1 0x5361c.12:1 => 0x00000001
sudo mstmcra mlx5_1 0x5363c.12:1 => 0x00000001
sudo mstmcra mlx5_1 0x53614.29:1 => 0x00000000
sudo mstmcra mlx5_1 0x53634.29:1 => 0x00000001
Modifying:
sudo mstmcra mlx5_1 0x5361c.12:1 0x0
sudo mstmcra mlx5_1 0x5363c.12:1 0x0
sudo mstmcra mlx5_1 0x53614.29:1 0x0
sudo mstmcra mlx5_1 0x53634.29:1 0x0
After:
sudo mstmcra mlx5_1 0x5361c.12:1 => 0x00000001
sudo mstmcra mlx5_1 0x5363c.12:1 => 0x00000001
sudo mstmcra mlx5_1 0x53614.29:1 => 0x00000000
sudo mstmcra mlx5_1 0x53634.29:1 => 0x00000001`

ogalbxela, 2024-07-18:
I consulted with the related engineer from our side. It appears that you are trying to alter values of the read-only registers. Behavior you observe is expected.

closed 2024-07-18 by ogalbxela

Eternity-Wang, 2024-07-19:
> I consulted with the related engineer from our side. It appears that you are trying to alter values of the read-only registers. Behavior you observe is expected.

Thank you very much for your reply and help. I have completed this test on top of another NIC.