Mellanox/mstflint #999: mft ? mstflint

xiaolongzhou123, 2024-09-28:
[image]
"MFT? MSTFlint
Are the source codes of these two different?

Unable to read switch-related information remotely, such as mlxlink -d lid-xxx -p 1/1."

xiaolongzhou123, 2024-09-28:
"MFT? MSTFlint
Are the source codes of these two different?

Unable to read switch-related information remotely, such as mlxlink -d lid-xxx -p 1/1."

xiaolongzhou123, 2024-09-28:
"How can I download the source code for MFT?
I need to remotely read switch content through an API."

ogalbxela, 2024-09-29:
mft and mstflint are not the same.

mft is proprietary product and only distributed in binary form https://network.nvidia.com/products/adapter-software/firmware-tools/

closed 2024-09-29 by ogalbxela