Mellanox/mstflint #1038: Usage of deprecated openssl "ENGINE API"

gobenji, 2024-11-28:
mstflint currently uses an openssl api that was deprecated since version 3.0, the "engine api":
https://github.com/openssl/openssl/blob/af33b200da8040c78dbfd8405878190980727171/README-ENGINES.md

This was pointed out by Red Hat employees in a private ticket (https://issues.redhat.com/browse/RHEL-58081). In RHEL-10, the next major release of RHEL, openssl is built without support for "engines".

In that case, trying to compile mstflint leads to errors starting with:
```
mlxsign_openssl_engine.cpp:86:5: error: 'ENGINE_load_builtin_engines' was not declared in this scope
   86 |     ENGINE_load_builtin_engines();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

As a result, mstflint is built without openssl and thus without mstfwmanager in RHEL-10. From mstflint.spec:
```
* Fri May 24 2024 Kamal Heib <kheib@redhat.com> - 4.26.0-4
- Disable openssl
```

Since mstfwmanager is the most convenient way to upgrade firmware, it would be nice to be able to have it in RHEL-10. Can you remove usage of the deprecated engine api in mstflint?

ogalbxela, 2025-01-25:
status here: requested change is implemented and should be part of upcoming 4.31 release

ich777, 2025-02-05:
@ogalbxela I see a Tag already for 4.31.0-1 but no Release, is there a release planned?

ogalbxela, 2025-02-06:
It is planned. The tarballs will be identical to what you can obtain from the v4.31.0-1 tag now.

ogalbxela, 2025-02-06:
References to the ENGINE API was removed starting https://github.com/Mellanox/mstflint/releases/tag/v4.31.0-1

closed 2025-02-06 by ogalbxela

gobenji, 2025-02-07:
For anyone coming here from RHEL-58081, while the openssl "ENGINE API" usage is removed in mstflint-4.31.0, that version cannot be compiled on RHEL-10.0 (#1133) so more work is still needed.