Mellanox/mstflint #1132: mstflint-4.31.0 fails to build from source on FC-41, error in mlxsign_lib.cpp

gobenji, 2025-02-07:
On a Fedora Core 41 system, doing 
```
dnf builddep mstflint
tar -xf mstflint-4.31.0-1.tar.gz
cd mstflint-4.31.0
./configure
make
```
fails because of the following error:
```
/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..  -I. -I.. -I../common -I../common  -Wall -W -g -MP -MD -pipe -fPIC -DTOOLS_CRYPTO_KEY='' -DTOOLS_CRYPTO_IV=''  -g -O2 -DHAVE_TERMIOS_H -DHAVE_SYS_PCI_H -DNO_RDMEM -DMST_UL -MT mlxsign_lib.lo -MD -MP -MF $depbase.Tpo -c -o mlxsign_lib.lo mlxsign_lib.cpp &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I. -I.. -I../common -I../common -Wall -W -g -MP -MD -pipe -fPIC -DTOOLS_CRYPTO_KEY= -DTOOLS_CRYPTO_IV= -g -O2 -DHAVE_TERMIOS_H -DHAVE_SYS_PCI_H -DNO_RDMEM -DMST_UL -MT mlxsign_lib.lo -MD -MP -MF .deps/mlxsign_lib.Tpo -c mlxsign_lib.cpp  -fPIC -DPIC -o .libs/mlxsign_lib.o
mlxsign_lib.cpp:40:10: fatal error: openssl/engine.h: No such file or directory
   40 | #include <openssl/engine.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:451: mlxsign_lib.lo] Error 1
```

ogalbxela, 2025-02-07:
There is a list of supported OSes in the release note. Fedora-41 is not officially suppored yet.

gobenji, 2025-05-15:
mstflint-4.32.0 builds on FC41

closed 2025-05-15 by gobenji